prusa3d/PrusaSlicer #10191: Ensure vertical shell thickness -Removal

Imtherealnoto, 2023-03-30:
### Description of the bug

Now that ensure vertical shell thickness has been removed these unwanted artifacts come up when printing single walled objects
[image]

### Project file & How to reproduce

see picture.
It worked fine before the Ensure vertical shell thickness option was removed.

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer 2.6.0-alpha5

### Operating system

Windows 10

### Printer model

MK3s+ MMU2

Tupson444, 2023-06-30:
This is another case when none of the extra material is needed. The "main" issue requesting ability to remove unwanted material is #10102

lukasmatena, 2024-11-21:
"Ensure vertical shell thickness" option is restored in [2.9.0-alpha1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.9.0-alpha1) and hopefully configurable enough to cover all use cases.  Closing.

closed 2024-11-21 by lukasmatena