prusa3d/PrusaSlicer #10192: Filament settings for individual parts in one print

Jogiarea, 2023-03-30:
Hello Prusa,

First of all thank you for providing your slicer
Just great.

As a small suggestion, what always makes me use Superslicer is that when printing multiple parts, you can't change the filament settings for each part separately.

I always like to use this to calibrate the flow and retract so you don't always have to restart each part when printing.

Maybe something like the Calibration section to automatically create calibration G-codes would be an idea, I have super calibration tools on the homepage, this directly in the slicer would be mega

Thanks again to you Prusa, that you support the Slicer so great and you the DIY Open Source memory remains.

[image]
[image]

ajshell1, 2023-05-19:
Just commenting to say that I really want this feature in PrusaSlicer as well. I think it's the biggest thing that I'm missing that's present in SuperSlicer.

myevo8u, 2023-06-01:
@Jogiarea  Well I decided to create a script for extrusion multiplier calibration. Check it out here: https://github.com/myevo8u/Prusa-Slicer-Extrusion-Multiplier-Calibration-Script

Zerufuru, 2023-06-25:
I am also seeking this feature, particularly for the ability to modify extrusion multiplier per object on the build plate (for tuning purposes, mainly).

inovatorius, 2023-12-14:
Looking forward for this feature as well!

closed 2026-09-29 by prusaslicer-bot[bot]