prusa3d/PrusaSlicer #10194: Organic supports generated with over 2 perimeters - allow control of number of perimeters

rlesac, 2023-03-30:
### Description of the bug

Organic supports are generated with more than one perimeter, almost doubling the amount of filament use.
I expected either to have one perimeter on supports or to have a setting to adjust the number of perimeters.
Only the thinnest branches have less than two perimeters, but never only one.

[image]
[image]

### Project file & How to reproduce

[prusaSlicer-bug-organic-support-perimeters-overhang-test-01.zip](https://github.com/prusa3d/PrusaSlicer/files/11110197/prusaSlicer-bug-organic-support-perimeters-overhang-test-01.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.6.0-alpha5+win64

### Operating system

Windows 11

### Printer model

Ender 3 running Klipper

YazvaArt, 2024-04-16:
I'm surprised this problem hasn't been solved by now.

neophyl, 2024-04-16:
Its called Branch Diameter with double walls.  If you only want a single perimeter then set it to a large value,
[image]

gives you 
[image]

You have it set to 3mm so any branch over 3mm gets double walls.

YazvaArt, 2024-04-16:
> Branch Diameter with double walls

Oh, thank you a lot!

closed 2026-09-29 by prusaslicer-bot[bot]