prusa3d/PrusaSlicer #10201: Dynamic Fan Speed Overrides Disable Fans for x Layers

jjz015, 2023-03-30:
### Description of the bug

When Dynamic Fan Speed is enabled, the fan turns on at layer 2 instead of whichever layer I had set the fans to turn on at in the slicer. 

### Project file & How to reproduce

[bugreport.zip](https://github.com/prusa3d/PrusaSlicer/files/11116465/bugreport.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0-alpha5

### Operating system

Windows 10

### Printer model

Voron 0.2

Sprutx, 2023-04-05:
I see the same thing on version 2.6.0-alpha6

If I install:

disable_fan_first_layers = 0
fan_always_on = 1
min_fan_speed = 100
max_fan_speed = 100
full_fan_speed_layer = 4

Fan not blowing on layer 1
It only turns on on layer 2

kubispe1, 2023-04-13:
2.6.0-alpha5
[image]

2.6.0-alpha6
[image]

In alpha6 seems to me it respects settings

@Sprutx Can you provide 3mf to check it in a6

Sprutx, 2023-04-13:
> In alpha6 seems to me it respects settings

It doesn't even work like that

[image]

[image]
[https://disk.yandex.ru/d/GvaeaFUgNpHNYw](url)

closed 2026-09-29 by prusaslicer-bot[bot]