prusa3d/PrusaSlicer #10202: Going back to 2.6 alpha 4

lucasavendano, 2023-03-31:
With 2.6 alpha 4 i couldn´t be more happy, but suddenly came alpha 5 with changes i didn´t expect because 9 bad comments on Ensure vertical shell thickness, for me it was perfect as it was, and now i have things that i can´t tweak because decided to eliminate Ensure vertical shell thickness switch....

[image]

What you show in the picture is cool but now PS wants to do that everywhere, i think it is not usefull for everything, for some things it will be usefull and for others not.

I can´t show where it is not usefull but i think as you did with "arachne" it would be nice that you give to us the posibility to choose if use concentric fill or not for ensure vertical shell thickness or in other places.

For now, going back to alpha 4 i can make it but... i can´t go back to 2.5, 2.6 will have a lot of cool stuff  and i can´t stay forever in alpha 4...

I was specting more a good work on travel path planning  than improve ensure vertical shell thickness... i think that if something works well for mayority... why touching it...

In other order of things, i am not sure but i think you did some other changes cause alpha 5 is like rough, now i have a printer skipping steps after travel/unretract...

I say "i am not sure" because with alpha 4 i installed klipper in all my printers, and in the process of calibrating everything came alpha 5 and problem started.

Today sliced with alpha 4 and all came to normality again.

Please don´t throw this comment to garbage i don´t want to use other slicer, i don´t like any of the othersss😁

Gunner087, 2023-04-02:
I personally like the changes since it removes all the small little back and forth movements. I have no idea how the algorithm actually works, but I think a size threshold or something must be set too high, causing concentric fill to be used in areas where it really doesn't need to be. The images below demonstrate this well, the first image is alpha 4 and the second image is alpha 5 with the same settings. In the case of this specific part I resorted back to using alpha 4 because the concentric fill caused more extreme curling at the sharp edge of the overhang.
[image]
[image]

Gunner087, 2023-04-02:
@lucasavendano it might help to draw attention to this problem if you revise the title to something that makes it easier to understand the issue, such as "Ensure vertical shell thickness applied in too many areas"

Tupson444, 2023-04-02:
This (or similar) is requested here: #10102

lucasavendano, 2023-04-07:
@Gunner087 ok, you are right.

lucasavendano, 2023-04-07:
> causing concentric fill to be used in areas where it really doesn't need to be

Hi, i think that this is mainly what i wanted to say.

As this improvement was presented in alpha 5 release i understood that it improves sloping surfaces, ok, good, but i encounter that it is being applied to other places with no slopes too or in small areas:

layer 2, 5x5x5 cube
[image]

[image]

In peaces like the ones in the pictures, the movement for making the concentric infill feels not much better than little zig zag´s

Maybe it is necessary reviewing the algorithm or enable the posibility of choose when to use it.

smajlySVK, 2023-04-07:
Hello, I noticed that new "Ensure vertical shell thickness improvement" affected top layer quality. 
On objects with width less than 6 mm concentric infill is applied and causes visibly areas which are worse than with rectilinear infill bellow the last layer.

I understand, that concentric infill can be printed faster, but it would be perfect to leave checkbox in user interface where user can select "old" or "new" infill behavior.

[image]

lucasavendano, 2023-04-11:
> Hello, I noticed that new "Ensure vertical shell thickness improvement" affected top layer quality.

Hi @smajlySVK, i think that it affects not only top layer quality but other places too. What you show in your pictures was the kind of parts that brought me here, but hoping this can be visible and reviewed i´ll leave other examples:

This was printed with PETG
[image]

In this particular place i needed only walls as shown below, so i put a modifier not for all height, just in middle sector
[image]

but just below that sector the new algorithm forces concentric infill like this:
[image]

As the concentric infil is very fast the part ended delaminated, someone would say "just print slower".. but i think that is not the way. So it is not suitable for every place, not to mention that in the example of toppers and in my pictures there are no slopes.

I ended up deciding to apply perimeters all the height all the part, when maybe it is not needed just because i cannot disable "ensure vertical shell thickness"
[image]

The result:
[image]

mmcglumphy, 2023-04-20:
Would like the ability to disable 'ensure' entirely, as many things that I print do not need extra internal support given enough perimeters.

Perhaps a threshold setting?

HardRock88, 2023-05-08:
I have this problem too, if model has walls thinner then 7.5 mm, then secondary layers looks like this:
[image]
this cause problems when I am printing something like this:
[image]
Printing is not consistent, it makes head move all around the model. 
In version 2.5.2 same model has Monotonic pattern for secondary layers, which is better I think:
[image]

lucasavendano, 2023-06-09:
hi, @kubispe1 i don´t want to be a pain for anybody but, this issue is attendable? could be a chance that in future releases the user can choose between concentric and traditional type of infill for the cases mentioned by @HardRock88 ?

All slicers based on PS adopted the same concentric infill in the same areas, i see it as a problem, but are we so few with this problem?

PirateOnGitHub, 2023-06-27:
Yes new infill is really bad , especially if you are using high temp filaments 240-250C.

saveman71, 2023-11-22:
Thanks to a [workaround from @neophyl on the printables.com forums](https://forum.prusa3d.com/forum/prusaslicer/removal-of-solid-infill-on-a-part-with-zero-infill/#post-681172), I was able to remove completely the solid infill by adding a height modifier on the whole height except bottom and top, then setting these parameters

[image]

But I would vastly prefer an option to turn that off in the settings.

closed 2024-11-21 by lukasmatena