prusa3d/PrusaSlicer #10205: 2.6 alpha 5 - 2 bridge infill layers after full infill layer

vpil, 2023-03-31:
### Description of the bug

Slicing my model in 2.6 alpha 5 using default setting (0.10mm DETAIL, Generic PLA, MINI & MINI+, 15% infill) puts 2 bridge infill (výplň mostů) layers - see 1.6 & 1.7 mm right after full infill layer (plná výplň). This does not happen in 2.5 (or even in 2.6 if I try to cut just a piece of my model) - there are correct full infill layers too.

### Project file & How to reproduce

[problem_2.6-a5.zip](https://github.com/prusa3d/PrusaSlicer/files/11120302/problem_2.6-a5.zip)
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0-alpha5

### Operating system

Windows 10 64bit

### Printer model

Mini

Jan-Soustruznik, 2023-04-04:
Hello, @vpil, 

could you check with last [Alpha6](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha6) release, please? I think that problem is solved. 

Thank you, have a nice day

vpil, 2023-04-04:
Thank you, I can confirm in alpha 6 it is OK.

closed 2023-04-04 by vpil