prusa3d/PrusaSlicer #10210: Internal error when using Mainsail/Fluidd physical printer

TheEnigmaBlade, 2023-03-31:
### Description of the bug

Attempting to use a physical printer with host type of "Mainsail/Fluidd" results in any attempts to connect to the printer failing.
- Pressing the "Test" button results in an "Unhandled unknown exception" and terminates the application with the error below.
- Uploading to the printer results in the error below but does not terminate the application.

[image]

[image]

I have tried both editing an existing physical printer functioning fine in Octoprint mode and creating a new physical printer.

### Project file & How to reproduce

1. Create or edit a physical printer.
2. Set printer type to "Mainsail/Fluidd".
3. Press "Test" or upload to printer.
4. Get error.

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0-alpha6+win64

### Operating system

Windows 11

### Printer model

Voron 2.4 w/ Mainsail

cpasoft, 2023-03-31:
Confirmed... I also have that error

VagnPetersen, 2023-03-31:
Confirmed... I also have same error.

iqubik, 2023-03-31:
also have this error

pauluchin, 2023-04-01:
I also have this error :(

joshua70448, 2023-04-01:
I'm also having the same error with Mainsail v2.4.1, using both host name and IP address.

Jonibygg, 2023-04-01:
Same here

Version of Prusa
2.6.0-alpha6+Mac Universal

Os
iMac MacOs Ventura 13.2

b4rRa, 2023-04-01:
100% exact the same issues with 2.6 alpha6

TravisWilder, 2023-04-01:
When I select Mainsail PS is setting up in the INI Duet.
It seems there are wrong offsets used

CloudYeti, 2023-04-01:
I have the same issue on MacOS

LazeMSS, 2023-04-02:
Confirmed

orion23rigel, 2023-04-02:
Same issue here

VagnPetersen, 2023-04-03:
For me it is working now.
Minor thing is that pressing the "test" button returns this: "Connection to OctoPrint works correctly"

Jonibygg, 2023-04-03:
> Thanks for the reports. Can you please try the following build and report back? It should be fixed.
> PrusaSlicer-2.6.0-alpha6+1-lm-mailsail-fix-win64-g9346cf119-202304030833 ***@***.***/files/lm/PrusaSlicer-2.6.0-alpha6+1-lm-mailsail-fix-win64-g9346cf119-202304030833.zip>
> PrusaSlicer-2.6.0-alpha6+1-lm-mailsail-fix-MacOS-universal-g9346cf119-202304030843.dmg

No working DL-link for MacOS in the mail?

TravisWilder, 2023-04-03:
> Thanks for the reports. Can you please try the following build and report back? It should be fixed.
> 
> [PrusaSlicer-2.6.0-alpha6+1-lm-mailsail-fix-win64-g9346cf119-202304030833](https://prusaslicer:slicer@slicerbuilds.prusa3d.com/files/lm/PrusaSlicer-2.6.0-alpha6+1-lm-mailsail-fix-win64-g9346cf119-202304030833.zip) [PrusaSlicer-2.6.0-alpha6+1-lm-mailsail-fix-MacOS-universal-g9346cf119-202304030843](https://prusaslicer:slicer@slicerbuilds.prusa3d.com/files/lm/PrusaSlicer-2.6.0-alpha6+1-lm-mailsail-fix-MacOS-universal-g9346cf119-202304030843.dmg) (No Linux build, sorry.)

MacOS (M1)
Setting changeable - test works .. ini is now set to "host_type = mainsail"

Jonibygg, 2023-04-03:
> PrusaSlicer-2.6.0-alpha6+1-lm-mailsail-fix-win64-g9346cf119-202304030833 ***@***.***/files/lm/PrusaSlicer-2.6.0-alpha6+1-lm-mailsail-fix-win64-g9346cf119-202304030833.zip> PrusaSlicer-2.6.0-alpha6+1-lm-mailsail-fix-MacOS-universal-g9346cf119-202304030843 ***@***.***/files/lm/PrusaSlicer-2.6.0-alpha6+1-lm-mailsail-fix-MacOS-universal-g9346cf119-202304030843.dmg> (No Linux build, sorry.)

What do I use here?

￼

GAZ082, 2023-04-03:
> Thanks for the reports. Can you please try the following build and report back? It should be fixed. [...]

Are we going to have a Linux release of this fix?

lukasmatena, 2023-04-03:
@TravisWilder and @VagnPetersen: Thanks.

@GAZ082 It will be fixed in the next 2.6.0 alpha/beta. You can use OctoPrint print host until then.

auditor3d, 2023-04-03:
**They have the drop down list out of order.  workaround until an updated alpha for linux...  use the Octoprint dropdown option and input your mainsail/klipper IP or hostname and test.**

churls5495, 2023-04-03:
> Thanks for the reports. Can you please try the following build and report back? It should be fixed. [...]

Win64 version fixed the issue for me.

tyrscgm, 2023-04-05:
Confirmed... I also have same error. with Mac osx Ventura

mpaw, 2023-04-07:
> Thanks for the reports. Can you please try the following build and report back? It should be fixed. [...]

Fixed the issue for me on MacOS Ventura (M1 Pro). Although the popup says `Octoprint` instead of `Mainsail/Fluidd`

[image]

pauluchin, 2023-04-07:
> Thanks for the reports. Can you please try the following build and report back? It should be fixed. [...]

No problem with this build Thank's

iqubik, 2023-04-07:
[PrusaSlicer-2.6.0-alpha6+1-lm-mailsail-fix-win64-g9346cf119-202304030833](https://prusaslicer:slicer@slicerbuilds.prusa3d.com/files/lm/PrusaSlicer-2.6.0-alpha6+1-lm-mailsail-fix-win64-g9346cf119-202304030833.zip)
**win10-64 all work correct**

cristianmur, 2023-04-09:
Todos tenemos el mismo error, lo implementan supuestamente para mejorar pero te fastida todo

pauluchin, 2023-04-09:
here everything works normally with the correction

GAZ082, 2023-04-10:
> Todos tenemos el mismo error, lo implementan supuestamente para mejorar pero te fastida todo

This has already been fixed. This is the release:
https://github.com/prusa3d/PrusaSlicer/issues/10210#issuecomment-1493976330

No linux version tho :(

e-dreyer, 2023-04-13:
Can confirm that this happens to me when I press the "test" button. Otherwise, the settings are saved and the same error occurs when I try to upload, and the PrusaSlicer crashes and closes. I am running Alpha 6

erwinsotiri, 2023-04-24:
When trying to upload I get: Error uploading to print host:
<unspecified file>(1): expected value

Darkknight33, 2023-05-08:
> 

I saw this issue when I had Octoprint selected as the destination in the A6 build. Test worked, however, would just say unresolved address on attempt to upload. Selecting Mainsail crashed the slicer. Using the testfix build, it works properly. Just uploaded a file.

lukasmatena, 2023-05-16:
Should be fine in 2.6.0-beta1. Closing.

closed 2023-05-16 by lukasmatena

GAZ082, 2023-06-12:
Working fine in linux.

Version: 2.6.0-rc1+linux-x64-GTK3-202306091240

Jan-Soustruznik, 2023-06-13:
> Problem is back again in 2.6.0-RC1+Win64-2002306091137

Hello, @TacII , could you share your 3mf file, please?

TacII, 2023-06-16:
Scratch that! I think the upgrade changed the default printer for me somehow. I think the same thing happend with RC2 now and I found the correct printer and it works now.

lukasmatena, 2023-06-16:
@TacII Ok, thanks for the heads-up.