prusa3d/PrusaSlicer #10219: When importing a config bundle, all presets are removed

dajomas, 2023-04-01:
### Description of the bug

I imported a config bundle provided by a filament producer (Real Filament) and afterwards, all my presets are gone (Print, Filament and Printer)

I expected the imported bundle to ADD the presets and not wipe existing presets

### Project file & How to reproduce

No project file was loaded when I imported the config bundle

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0-Alpha6

### Operating system

Windows 11

### Printer model

Creality Ender 5 Plus

neophyl, 2023-04-01:
So any chance of you actually attaching the config bundle that wipes all your profiles ?  At least give the devs a slight chance of being able to debug your problem.  How do you expect them to be able to even start without the problem file ?

dajomas, 2023-04-01:
[PrusaSlicer_config_bundle.txt](https://github.com/prusa3d/PrusaSlicer/files/11129643/PrusaSlicer_config_bundle.txt)

I have attached the ini file but had to rename it to a txt file. The file was originally downloaded from here: https://real-filament.com/techspecs

closed 2026-09-29 by prusaslicer-bot[bot]