prusa3d/PrusaSlicer #10220: Open STL file with German umlauts broken

Acidham, 2023-04-01:
### Description of the bug

When double-click on .stl file contaiing a german umlaut PS 2.5.2 start empty gcode viewer instead of PS workbench. When removing german umlaut it works as expected

### Project file & How to reproduce

1. ensure .stl file is assigned to open with PrusaSlicer
2. double-click on .stl file with at least one German umlaut, e.g. `körper.stl`
3. Prusa G-Code viewer will be opened instead of PrusaSlicer
4. change filename from `körper.stl` to `koerper.stl` 
5. double click to open
6. PS will be opened properly.

Two stl models: [Archive.zip](https://github.com/prusa3d/PrusaSlicer/files/11129043/Archive.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.2-Macos-ARM64

### Operating system

macOS 13.3

### Printer model

Prusa MK3s+

Olafejs, 2023-04-03:
In PL names the same.

Acidham, 2024-02-25:
Seems to be fixed

closed 2024-02-25 by Acidham