prusa3d/PrusaSlicer #10221: strange skipping of infill

kdyrja, 2023-04-01:
### Description of the bug

prusa slicer strange skipping of infill

[image]

### Project file & How to reproduce

default MK3s settings, round object and infill 60%

[BUG_1_0.2mm_PETG_MK3S_1h34m.zip](https://github.com/prusa3d/PrusaSlicer/files/11129320/BUG_1_0.2mm_PETG_MK3S_1h34m.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.1

### Operating system

windows 11

### Printer model

mk3s+

neophyl, 2023-04-01:
Without a proper prusa project file as requested when opening an issue its going to be impossible to debug anything.  
Gcode is not very useful its all the settings you have as well as the object that matter.  Even placement etc could possibly make a difference.

kdyrja, 2023-04-01:
> Without a proper prusa project file as requested when opening an issue its going to be impossible to debug anything. Gcode is not very useful its all the settings you have as well as the object that matter. Even placement etc could possibly make a difference. Which default Mk3 settings ? Maybe you accidentally have only infill where needed selected for example. Without a project whi can tell.

As I wrote. Default factory settings, any cylindrical shape and infill at 60%

[Shape-Cylinder.zip](https://github.com/prusa3d/PrusaSlicer/files/11129718/Shape-Cylinder.zip)

closed 2026-09-29 by prusaslicer-bot[bot]