prusa3d/PrusaSlicer #10222: Lightning Infill causes infintely long bridges due to extended infill

Ockenfuss, 2023-04-01:
### Description of the bug

Lightning infill can cause infinitely long bridges inside the model, when applied on top of another infill type.

This is likely due to the new extended sparse infill, introduced in PrusaSlicer2.6.0alpha5. According to the Alpha5 release notes, the new extended infill is not applied if Lightning infill is chosen, but obviously this rule breaks when we introduce a height range modifier with another infill type. As a result, the sparse infill lines are extended until they reach a regular infill line or the model wall, which can be located far away in case of lightning infill.

### Project file & How to reproduce

Create a simple model with a depression in the middle, e.g. a cube with a smaller cube subtracted on top. Slice it with lightning infill, but apply another infill type (e.g. grid) at the bottom. Take a look at my "Box" project for an example.

Originally, I was trying to slice a landspace model with an open, gridded bottom and lightning on top. There, the issue appears even stronger.

[MountainRelief_LightningBridgeExample.zip](https://github.com/prusa3d/PrusaSlicer/files/11129394/MountainRelief_LightningBridgeExample.zip)
[Box_LighntingBridgeExample.zip](https://github.com/prusa3d/PrusaSlicer/files/11129397/Box_LighntingBridgeExample.zip)

[image]
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.6.0-alpha6+linux-x64-GTK3

### Operating system

Ubuntu 22.04.2 LTS

### Printer model

Prusa MK3S

Jan-Soustruznik, 2023-04-03:
Issue created SPE-1628

kubispe1, 2023-05-18:
Changelog notes of PS 2.6.0-beta1: 
`Improved the algorithm so it does not generate extremely long bridges when the layer below is very sparse.`
Thanks

closed 2023-05-18 by kubispe1