prusa3d/PrusaSlicer #10223: Organic Supports are not fully supporting with Alpha 6, with Alpha 5 they did.

gregshawdj, 2023-04-01:
### Description of the bug

I have just tried PrusaSlicer Alpha 6 with a Toad egg model and it looks like something has changed for the worse when it comes to organic supports.

Here I have the exact same .3mf file with the same support points sliced with Alpha 5 and Alpha 6. As you can see Alpha 5 does a much better job of having supports ready for the overhang, whereas Alpha 6 have gaps which causes some of the overhangs to be printed in mid air.

Doing a side by side comparison of the support settings (Alpha 6 on the left and Alpha 5 on the right) there is a new setting called branch distance in the organic supports section. I will try some different values to see if I can get what Alpha 5 gives because currently Alpha 5 has the better supports.

Reading the docs that came with the update it mentions 'Organic supports: reduced default support_tree_top_rate to 15%' However this is not a visible setting to change or searchable within the software.

I've added screen shots showing the difference and screen shots using Alpha 6 where you can see an island being printed with no supports underneath. With Alpha 5 there were supports underneath.

### Project file & How to reproduce

[Hex3D_Toad_Egg_Mono.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/11129696/Hex3D_Toad_Egg_Mono.3mf.zip)

[image]
[image]
[image]
[image]
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0 alpha 6

### Operating system

MacOS 12.6.3

### Printer model

Creality Ender 5 Pro

Braintoe, 2023-05-06:
I second this (2.6.0 alpha 6 on Linux). For a part that has a radius on the inside going from vertical to almost horizontal, there is no change in the area that is supported with organic supports for a support material threshold value of 80° and 40°, whereas a normal support varies significantly and starts to support overall earlier.

support material threshold value 80° - organic support: support begins to contact part on layer 83:
[image]

support material threshold value 80° - standard support: support begins to contact part on layer 28:
[image]

support material threshold value 40° - organic support: support begins to contact part on layer 83 as well:
[image]

support material threshold value 40° - standard support: support begins to contact part on layer 75:
[image]

Support material contact distance is set to 0 (to be printed with a dedicated support material), support material XY spacing is set to 0,3mm.

Please find the part I used attached. Used layer height in the example was 0.15mm.
[schale_deckel.stl.zip](https://github.com/prusa3d/PrusaSlicer/files/11410817/schale_deckel.stl.zip)

Braintoe, 2026-09-06:
Still interested in that and still unsolved, therefore adding this comment on order to avoid closing the issue.