prusa3d/PrusaSlicer #10224: Scale factors table possibly showing incorrect values when using support blocker modifier

Firefighter1023, 2023-04-01:
### Description of the bug

When I have objects on the build plate, single or multiple, once a support blocker is added, the scale factors on the right hand side will present conflicting information on what the actual scale (percent) is of the original object (STL).

In this first screen shot, the object is scaled to 90% of the original value, however when adding the support blocker it is now reported as 135% respectfully in World Coordinates:
[image]

When selecting the object in the file manager (right table), and switching to world, object or part coordinates, it reports that the object is at 100% scale, which is also incorrect as it should be reporting 90% scale.
[image]
[image]
[image]

In previous versions, <2.5, this conflict did not occur, only in 2.6 alpha versions.

### Project file & How to reproduce

Screenshots are included inline with the message.

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6a6

### Operating system

Windows10

### Printer model

MK3S+/Artillery Sidewinder X1/X2

timur-tabi, 2024-11-16:
This problem still exists in PrusaSlicer 2.8.1.

Firefighter1023, 2024-11-16:
> This problem still exists in PrusaSlicer 2.8.1.

Agreed, I’m still finding that this same problem exists.  It’s been over a year since I reported it, I don’t have much hope that it will get any attention.

timur-tabi, 2026-09-03:
This should still be fixed.