prusa3d/PrusaSlicer #10226: Organic support prints higher than print itself, print crashes, and nozzle melts onto print 3 hours into print

b4de4nd, 2023-04-01:
### Description of the bug

So print is based with organic support for an overhang at 75mm from build plate. Support looks okay in slicer

[image]

However on the right side of the print, support was so filled up that the Pinda crashed into it and took the model of the bed, but not more than the nozzle could dig in, an the print was almost turning to a blob. Lucky for me i was sitting right next to the machine and saw it happening.

I have had a lot of random crashes on prints on alpha versions, but ive never managed to tell why before today where i saw the support being build higher than the print itself.

[Prusa c920 Camera Y Axis v4.zip](https://github.com/prusa3d/PrusaSlicer/files/11130302/Prusa.c920.Camera.Y.Axis.v4.zip)

### Project file & How to reproduce

[Prusa c920 Camera Y Axis v4.zip](https://github.com/prusa3d/PrusaSlicer/files/11130304/Prusa.c920.Camera.Y.Axis.v4.zip)

Printed from file through octoprint 1.8.7 stable version

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0 Alpha 6

### Operating system

Windows 10

### Printer model

i3mks+ with revo six

kubispe1, 2023-04-02:
Hope I understand clearly... That's quite common, cause in settings there is a `Top contact Z distance 0.2`  in support settings (to separate support and model).
You can decrease it to 0.1 or even use 0 - for soluble support. Do you mean this?
More or less the nozzle should not crash into the model....

b4de4nd, 2023-04-02:
HiNo the print is not the top Z distance. The settings are 0.2 and they work fine, however it printed the support higher than the print itself, so the supports would get higher than the model and therefore the nozzle hits the support. Before it Melted into the model it looked like the support was somewhat ingrown to the model, and the print suddenly contained support materiel on that side as well. The crash happened after the layers for the top part was done and it was no longer during the bridgepart of overhang.Regards Kenth

b4de4nd, 2023-04-03:
[image]

On the right side the model had filament as High as 1mm before the crash, unfortunaly the image is hard to tell from since the nozzle melted most of it Down. 

But it happened on top of the layers that was made on top of the organic support. Hope this makes sense :)

kubispe1, 2023-04-05:
I also try to print.  Seems to me ok. But let's open the issue. I inspected also the travels and toolpath. I cannot find any problem that caused the nozzle to crash. Have you tried to print it multiple times or just once with this result? 
Hope I did not overlook something... 
[image]
[image]

jamincollins, 2023-05-12:
I believe I'm also being hit by this, while printing with organic supports I can sometimes hear an impact with part of the model and periodically find a tree support has failed.

jamincollins, 2023-05-13:
I managed to capture video of 4 supports being knocked down by print head impacts.  The first two happen shortly after the beginning of the video. The head move is due to taking a timelapse frame.  The third happens around 6:43 into the video, with a head move that should be part of the generated gcode. The fourth happens around 8:04 and I believe it is the clearest example as the video even shows the print head jump from the impact.

Sharing via Google Drive as the video is ~345M:
https://drive.google.com/file/d/1GzxYO_MI3YeJvioT_kAL4hD4yzM80hCl/view?usp=share_link

istvanovics, 2023-05-15:
Same problem here guys! After everything was fine-tuned on my printer this problem happening with every print with organic support. I tried a lot of options but unfortunately, the problem is still there.

jamincollins, 2023-06-03:
This _might_ be fixed by the changes in [2.6.0-beta4](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-beta4).  Specifically:
> PrusaSlicer now performs a check for colliding toolpaths after slicing. This prevents print failures when multiple objects (or their supports) intersect and would be printed one over the other (e.g. https://github.com/prusa3d/PrusaSlicer/issues/316, https://github.com/prusa3d/PrusaSlicer/issues/10148 and many others).

jamincollins, 2023-06-09:
Can confirm the problem still exists, reprinted the same model as before and several supports were broken when the print finished.

jamincollins, 2023-10-03:
I'm still experiencing this on 2.6.1, as demonstrated in the attached image:
[image]

nicole-ashley, 2023-10-11:
This happens all the time for me. I've had to start using Cura for large support-heavy prints because it has the option to travel around printed objects, and I was sick of pausing the print, finding the broken pieces (often scattered across the room due to the speed they were broken off) and trying to glue them back on in exactly the right place ...

idrilirdi, 2024-02-14:
Yes, this is happening to me too right now with a print that's heavy in organic supports. As I have z-hop deactivated and the supports are printed first, when doing travel moves the nozzle loudly hits the supports sometimes. This should really be fixed

Eberek, 2024-06-19:
This is happening to me even with the normal support

freakydude, 2024-06-19:
What seems to help is to enable zhop - lift / ramp as a workaround (at least it helped me).
But it does not solve the problem. 

The slicer should avoid going directly to the support structures (or better through any kind of fragile structures like support trees) or in case of the support maybe it's easier to implement to finally print the support per layer.

Is there any comment from prusa dev team about this or similar issue out there?

closed 2026-09-29 by prusaslicer-bot[bot]