prusa3d/PrusaSlicer #10227: Arrange is putting one object too close to another object

timur-tabi, 2023-04-01:
### Description of the bug

In this one particular instance, the Arrange feature has placed one object touching another object, despite the gap distance of 6mm.  I can move the object away manually, press Arrange again, and it will just move back.

[image]

### Project file & How to reproduce

[nutjob.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/11130465/nutjob.3mf.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.5.1+linux-x64-GTK3

### Operating system

Ubuntu 22

### Printer model

MK3s

kubispe1, 2023-04-02:
You are right, 2.5.1 and 2.5.2 do not work well for your model. 
When I use 2.6.0-alpha6 and repair model with defect. It arranges OK. Let's try on Linux a6 please. 
[nutjob_repair.zip](https://github.com/prusa3d/PrusaSlicer/files/11133248/nutjob_repair.zip)
[image]

timur-tabi, 2023-04-02:
Confirmed.  It looks like 2.6 has a different arrangement implementation so it produces a very different result.

timur-tabi, 2023-04-02:
I spoke too soon.  I see a similar problem with 2.6-alpha6:

[image]

closed 2026-09-29 by prusaslicer-bot[bot]