prusa3d/PrusaSlicer #10229: Overhang Perimeters use Travel speed and ignore limits

undoz, 2023-04-02:
### Description of the bug

Overhang perimeters feature use Travel speed and ignore any machine or flow limits

Tested with:
Both Dynamic overhang speed on or off
Extra perimeters on overhangs off

[image]

### Project file & How to reproduce

[overhang speed bug.zip](https://github.com/prusa3d/PrusaSlicer/files/11131069/overhang.speed.bug.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.6.0-alpha6

### Operating system

Windows 10

### Printer model

MK3S, etc

undoz, 2023-04-02:
Traced the bug to enabling dynamic fan speeds in the Filament section

treyd3, 2023-04-03:
Came to write this up and found existing bug. Can confirm what is stated above: slice is fine till enabling Dynamic Fan Speeds. Then (some) overhanging perimeters use travel speeds.

[image]

rtyr, 2023-04-04:
Reproduded. Thanks.

Godrak, 2023-04-04:
The issue has been found and it will be fixed in the next release.

closed 2023-04-17 by FidelCapo