prusa3d/PrusaSlicer #10230: idle_temperature default value too high in 2.6.0-alpha6

TypQxQ, 2023-04-02:
### Description of the bug

The default value for Filament Settings idle_temperature is unusable hight (2147483647).
This should be defaulted to "temperature".

### Project file & How to reproduce

Do not activate for a filament "Idle trmperature" but use it in the macros, for example:
M568 P[initial_extruder] R{idle_temperature[initial_extruder]} S{first_layer_temperature[initial_extruder]} A1

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.6.0-alpha6+win64-202303311519

### Operating system

Windows 11 (build 22621), 64-bit edition

### Printer model

Custom Toolchanger

apphead, 2023-04-03:
I disagree. What would be the point of the idle_temperature if it defaults to temperature?

I think, the way prusa implemented it makes sense.  Don't use idle_temperature  if default value is returned. In the script below, I assume that any temp below 1000 is a valid idle_temperature , otherwise use standby_temperature_delta. Could not find a combination of idle_temperature  and standby_temperature_delta settings for multi-material prints that returned a "bad" value.

```
{if ooze_prevention==true}
	{if idle_temperature[current_extruder]<=1000} ;default value for idle_temperature is 2147483647, used if  idle_temp is deactivated
	M109 S{idle_temperature[current_extruder]} C3 W1 ;set filament idle_temp T{current_extruder}	
	{else}
	M109 S{temperature[current_extruder] + standby_temperature_delta} C3 W1 ;set print ooze_temp T{current_extruder}
	{endif}
{endif}

```

lukasmatena, 2023-04-03:
The idle-temperature option is "checkable". In the UI, there is a checkbox that defines whether that option is currently used or not. You generally need to check whether it is enabled and only use when it is. There is a function `is_nil` which will provide that information.

apphead, 2023-04-03:
How would you check if the idle_temperature is checked? Could not find a parameter name...

apphead, 2023-04-05:
@lukasmatena Thanks for your suggestion. It works as intended.

closed 2026-09-29 by prusaslicer-bot[bot]