prusa3d/PrusaSlicer #10231: Bridge do not attach perimeter, end just drops

rinkek, 2023-04-02:
### Description of the bug

When printing bridges, the end just drops without attaching to the perimeter. This works fine for the first two passes of the bridge, but not for the infill.
It seems that the parameter "Infill/perimeters overlap" doesn't do anything.
[image]

### Project file & How to reproduce

Print the bridge file and see the bridge dropping
Or just upload the file, slice and look at the ends of the bridge.

[Bridge Test.zip](https://github.com/prusa3d/PrusaSlicer/files/11131623/Bridge.Test.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0 alpha 6

### Operating system

Windows 10

### Printer model

Mini+ (bear)

Itox001, 2023-04-02:
Looks similar to this issue I reported on aplha5: #9988

kubispe1, 2023-04-03:
Hi, there is .3mf with your issue for test and investigation. 
[Working_Mini_Bridge_test.zip](https://github.com/prusa3d/PrusaSlicer/files/11135690/Working_Mini_Bridge_test.zip)

rinkek, 2023-04-03:
> Looks similar to this issue I reported on aplha5: #9988
Yes it does, i will try to see if the benchy behaves the same.

rinkek, 2023-04-03:
> > Not sure what you mean, this is the STL i designed, which seems to work fine.
> > The problem is in the other file. It fails every time.
> 
> Yes, everything is ok. It is for our purpose... 3mf with exact settings is for us much beneficial, so I attached it directly.

Ah, makes sense. But you have to realize that that one is working somehow. The other STL is failing.
Cheers

rinkek, 2023-04-03:
tested with a normal standard benchy. 
The first layer of the deck is also processed not correctly.
It seems the bridge is not attached properly
[image]

vovodroid, 2023-04-11:
I have similar issue #10313 , in case trigger was extrusion width.

rinkek, 2023-04-18:
> I have similar issue #10313 , in my case trigger was extrusion width.

I tried it with the default settings, still have this issue.

FidelCapo, 2023-04-19:
The original issue from @rinkek is fixed. Issue https://github.com/prusa3d/PrusaSlicer/issues/10313 is a different issue and is not fixed yet.

crx-cooper, 2023-05-14:
Just tried **2.6.0 beta 1**. 
The issue is still present. Bridge won't connect to perimeter:

[image]

Have tried setting extrusion-width to default and played with other settings to no avail.

Project File: 
[bridging_issue.zip](https://github.com/prusa3d/PrusaSlicer/files/11470879/bridging_issue.zip)

gojux, 2023-05-18:
I think I have a similar issue running 2.6.0 beta2. I have very thin walls where the bridging starts. With 2.5.0 and older versions these bridges had a small overlap with the perimeter.

2.5.1:
[image]

2.6.0 beta 2:
[image]

Failed part:
[image]

gojux, 2023-06-05:
It is better now, but the first line starts in the air and drops:
[image]

isaacsgraphic, 2023-06-27:
I'm getting the same issue I think as gojux. Prusa 2.6.0, bridge lines are just barely touching the perimeter, and not anchoring properly at all. It would be cool to get a setting to improve bridge overlap/anchoring. 
[image]

Toffypops, 2023-07-28:
Struggling with a similar problem trying to print a Temp Tower with pre-spliced profile, i see the innfill/wall overlap for me is set til 15% for bridges. The anchorlines print just fine, but then it anchors the first point, then misses the 2nd anchor point and the rest is just a blobb of horrible mess. 

Found the setting in Orca Slicer, called Innfill/wall overlap, which i think might fix the problem. Havent tested printing, but I would assume OP's problem here might be that his innfill is set to something like 0 or lover than 15% which i have. 

Not sure if your slicer has a similar setting, but i would think so. 
[image]

What would be nice was if there was a possiblity to input a custom gcode at a specific layer for the innfill. maybe you can i dunno. 

Enclosed is a picture showing my setting which i will try next.

Gallion, 2023-08-09:
I get a similar issue when the bridge is 1 layer thick (i.e. when running tests). The anchor doesn't extend into the print.

[image]

The top bridge is 2 layers thick at 0.2mm per layer, 0.4mm total thickness.
The bottom bridge is 1 layer thick at 0.2mm.

kubispe1, 2023-08-22:
We have improved region detection in task SPE-1726. This should eliminate some situations where thin regions separate bridges. However, there may still be models with slicing parameters where the issue could appear. Please keep this in mind. I am closing this issue for now.

To @Toffypops and @Gallion: There is no 3mf to check, but your issue probably refers to a different problem. Please try the upcoming PS 2.6.1-rc1 and let us know if it fixes the issue.

closed 2023-08-22 by kubispe1

Aerospacesmith, 2023-09-21:
I'm still getting this issue in 2.6.1 Release. I have to turn Infill Overlap to over 100% to get bridging to just barely connect to the perimeters, but of course then regular infil overlaps way too much. Anything less than 100% overlap and my bridges do not connect.
2.5.0 didn't have this issue and bridges anchored properly.

Vadforean, 2023-09-27:
[image]
[image]
I can confirm that the issue is still there in 2.6.1. I opened the same project in several versions of PrusaSlicer and the 2.6-versions "turn around" just before reaching the square tower in this bridging test model.
* It's the same on both sides of the model.
* Rotating the model 90° around Z doesn't change the behaviour.
* Setting Infill Overlap to 100% did not change this at all for me.

I also started 2.6.1-rc2 and sliced it there. It looks exactly like 2.6.1.

FlintEastwood, 2023-10-22:
Please reopen this issue!
Bridges starting in the air. "Top solid infill" is generated between the bridging parts, but it is not possible to print that.
Bridges will fall down!!!
Please see #11500 .
Last working version is v2.5.2 .
[image]
[image]
[image]

Vadforean, 2023-10-23:
Yes, I agree! Please reopen this, or handle it in #11500, which has now also been closed. In that issue, it seems version 2.7a has also been tested, with the same result.

To see the problem, just open a bridging test model and compare the slicing output from 2.5.2 with any version from 2.6.0 up. It is not printable and the bridge will drop in one or both ends. Whenever I have a model containing any bridge, I must now go back and slice that in 2.5.2 instead of the latest version, for the bridges not to fail.

Strilock, 2023-10-29:
Can confirm still happens in 2.7 alpha as well. This is really bad, it causes failed bridges and failed prints. There either has to be a setting to control the overlap or ideally the bridges cannot be printed mid air and part of international perimeter has to done via bringing. 
[image]

zbrozek, 2023-11-20:
Any word on this? Bridges still appear to be broken in 2.7.0-rc1.