prusa3d/PrusaSlicer #10234: PrusaSlicer-2.6.0-alpha6 Detect Thin Walls is not selectable

zoell, 2023-04-02:
### Description of the bug

When I open Advanced settings I cannot select Detect Thin Walls in the Layers and perimeters page.
The icon next to it is yellow and the lock is locked. I try to click the checkbox but it won't enable it.

### Project file & How to reproduce

1. Select Advanced mode
2. Go to Print Settings
3. Click Layers and Perimeters
4. Try to enabled Detect Thin Walls

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.6.0-alpha6

### Operating system

Windows 10

### Printer model

N/A

peterjhl, 2023-04-02:
From the PrusaSlicer FAQ/Help/WIKI.

> Detect thin walls

By default, every wall consists of outside and inside perimeters (a minimum of two perimeters for thin walls). If there is enough space for it, an infill pattern is used to fill the void between these inside/outside perimeters. 

Enabling Detect thin walls lets PrusaSlicer generate just a single perimeter that works as both inside & outside shell. This will help to capture tiny details. However, walls thinner than a single perimeter extrusion width will likely still be ignored.

**_This feature is automatically disables, if the Arachne perimeter generator is active, as it already automatically adjusts for thin walls_**. <    [https://help.prusa3d.com/article/layers-and-perimeters_1748#detect-thin-walls](url)

rtyr, 2023-04-02:
Answered. Closing.

closed 2023-04-02 by rtyr