prusa3d/PrusaSlicer #10238: Improve multimaterial painting color borders in concave geometry

jb221, 2023-04-02:
### Description of the bug

Often times models have concave geometric borders where colors would split, however PrusaSlicer seems to waffle back and forth on which color should get "right of way" causing very extreme visible artifacts in the color border where the model has a very smooth border.

Attached are 3 screenshots, SS1 shows what the color regions are in 3d editor mode, SS2 shows the same portion of the model in preview mode, and SS3 shows a layer cutoff that exposes the cause, the internal geometry of the color region having no consistency from layer to layer.
[image]
[image]
[image]

### Project file & How to reproduce

This particular file I can't share but it happens very frequently with concave geometric borders that are organically modeled and something I've struggled to work around on many projects. It seems that minor changes in the angle of the geometry at the border can play a huge role in where the physical separation is placed on the model, and if there was some way to force consistency between layers then it might help this issue.

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0

### Operating system

Windows 10 64bit

### Printer model

Prusa MK3S+

kubispe1, 2023-04-03:
I saw this issue already, but still, can you attach your .3mf? Here is an issue #8745 with a similar result but probably caused by a different reason.

jb221, 2023-04-03:
> I saw this issue already, but still, can you attach your .3mf? Here is an issue #8745 with a similar result but probably caused by a different reason.

I've added a cut down version of the file that still doing it, there are fewer colors but the issue is still present so it should work.

kubispe1, 2024-03-14:
Thanks, we selected this issue to upcoming plan of futures releases
SPE-2194

lukasmatena, 2025-02-18:
Should be fixed in [PrusaSlicer 2.9.1-alpha1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.9.1-alpha1). Closing.

closed 2025-02-18 by lukasmatena