prusa3d/PrusaSlicer #10239: {print_time} in file name

spuder, 2023-04-03:
### Description of the bug

Slicing a file in version 2.6.0-alpha5 results in a save dialog box with the string literal `{print_time}`.
I've reproduced this multiple times

1. open a STL file (I used the Egypt vase model from [here](https://www.printables.com/en/model/412796-pre-dynastic-egyptian-vase/files))
2. Slice
3. Click 'export gcode' in the bottom right. 

### Project file & How to reproduce

Download this file and slice it for any printer

https://www.printables.com/en/model/412796-pre-dynastic-egyptian-vase/files

[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0-alpha5

### Operating system

Mac 12.5.1 (M1 CPU)

### Printer model

Ender 2 pro

rtyr, 2023-04-03:
You have to wait till slicing is finished before exporting the g-code.

spuder, 2023-04-03:
Thanks. I thought I tested that edge case. 

I just tried again and confirmed I no longer see the error.

closed 2023-04-03 by spuder