prusa3d/PrusaSlicer #10241: Crash with "Arrange" and no printable items

MattBlissett, 2023-04-03:
### Description of the bug

When opening a project with only non-printable objects, clicking the "Arrange" button causes a crash.

### Project file & How to reproduce

[test.zip](https://github.com/prusa3d/PrusaSlicer/files/11137795/test.zip)

Steps:
1. New project
2. Add a cube
3. Change it to non-printing
4. Save project
5. Exit
6. Open project
7. Click arrange — crash!

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0-alpha6

### Operating system

Ubuntu 22.04

### Printer model

Prusk Mk3S

foreachthing, 2023-04-03:
My PS does not crash - on Windows 10.

MattBlissett, 2023-04-03:
I have a 620MB core dump, with a backtrace:

```
Core was generated by `/tmp/.mount_prusa-kL6peR/usr/bin/bin/prusa-slicer test.3mf'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x00000000007aaba7 in ?? ()
[Current thread is 1 (LWP 1410574)]
(gdb) backtrace
#0  0x00000000007aaba7 in ?? ()
#1  0x0000000002dcc4d8 in ?? ()
#2  0x3fdae147ae147ae1 in ?? ()
#3  0x000000000824f700 in ?? ()
#4  0x0000000000000000 in ?? ()
```

but I don't know if that's any use.

foreachthing, 2023-04-03:
Can you start PS with `prusa-slicer-console --loglevel=9` on Linux?

MattBlissett, 2023-04-03:
I am running the AppImage, but I created a symlink `prusa-slicer-console -> PrusaSlicer-2.6.0-alpha6+linux-x64-GTK2-202303311518.AppImage` in case that helps. I have a similar link `prusa-gcodeviewer -> PrusaSlicer-2.6.0-alpha6+linux-x64-GTK2-202303311518.AppImage` and `prusa-gcodeviewer` does launch into the G-code viewer.

<details>
  <summary>Initial output</summary>
<pre>
> prusa-slicer-console --loglevel=9 test.3mf
[2023-04-03 14:31:21.276044] [0x00007f81a2b18880] [trace]   Initializing StaticPrintConfigs
[2023-04-03 14:31:21.425026] [0x00007f81a2b18880] [trace]   App updater default download path: "/home/matt/"
[2023-04-03 14:31:21.425896] [0x00007f81a2b18880] [debug]   full path: 853089106416610175
[2023-04-03 14:31:21.425906] [0x00007f81a2b18880] [debug]   single instance: undefined. other params: /tmp/.mount_prusa-KU7u2h/usr/bin/bin/prusa-slicer;--loglevel=9;test.3mf
[2023-04-03 14:31:21.425911] [0x00007f81a2b18880] [debug]   full lock path: /home/matt/.config/PrusaSlicer-alpha/cache/853089106416610175.lock
[2023-04-03 14:31:21.425930] [0x00007f81a2b18880] [debug]   Not creating lockfile.
[2023-04-03 14:31:21.425934] [0x00007f81a2b18880] [info]    Instance check: Another instance not found or single-instance not set.

(prusa-slicer:1420419): Gtk-WARNING **: 14:31:21.466: Unable to locate theme engine in module_path: "adwaita",
[2023-04-03 14:31:21.468949] [0x00007f81a2b18880] [trace]   System language detected (user locales and such): en_IE
[2023-04-03 14:31:21.477476] [0x00007f81a2b18880] [trace]   Best translation language detected (may be different from user locales): en
[2023-04-03 14:31:21.478695] [0x00007f81a2b18880] [trace]   Switching wxLocales to en
[2023-04-03 14:31:21.685256] [0x00007f81a2b18880] [info]    Checking if indices need to be installed from resources...
[2023-04-03 14:31:22.346102] [0x00007f81a2b18880] [info]    Undo / Redo snapshot taken: New Project, Undo / Redo stack memory: 0MB Resident memory: 170MB; Shared memory: 92MB; Private memory: 78MB; Peak memory usage: 170MB
[2023-04-03 14:31:22.347649] [0x00007f8166ffd640] [error]   Not primary owner of DBus name - probably another PrusaSlicer instance is running.
[2023-04-03 14:31:22.347664] [0x00007f8166ffd640] [error]   Dbus Messages listening terminating.
[2023-04-03 14:31:22.401507] [0x00007f81a2b18880] [debug]   BuildVolume bed_shape clasified as: Rectangle
[2023-04-03 14:31:22.934443] [0x00007f81a2b18880] [trace]   'Noto Sans 10'font descriptor string param of load_wxFont()
[2023-04-03 14:31:22.934460] [0x00007f81a2b18880] [trace]   'Noto Sans 10' wx string descriptor
[2023-04-03 14:31:22.934519] [0x00007f81a2b18880] [trace]   loaded font is 'Noto Sans'.
[2023-04-03 14:31:22.957405] [0x00007f81a2b18880] [trace]   'Noto Sans 10' wx string get from GetNativeFontInfoDesc. wxFont IsOk(true), isNull(false)IsFixedWidth(false), IsUsingSizeInPixels(false), Encoding(43), 
[2023-04-03 14:31:23.024743] [0x00007f81a2b18880] [debug]   TriangleMesh::repair() started
[2023-04-03 14:31:23.025614] [0x00007f81a2b18880] [debug]   TriangleMesh::repair() finished
[2023-04-03 14:31:24.103406] [0x00007f81a2b18880] [info]    Undo / Redo snapshot taken: Load Project: test, Undo / Redo stack memory: 0MB Resident memory: 645MB; Shared memory: 132MB; Private memory: 513MB; Peak memory usage: 756MB
[2023-04-03 14:31:24.198302] [0x00007f81a2b18880] [info]    Checking for cached configuration updates...
[2023-04-03 14:31:24.198335] [0x00007f81a2b18880] [info]    Confing bundle not installed for vendor Artillery, skipping: 
[2023-04-03 14:31:24.198349] [0x00007f81a2b18880] [info]    Confing bundle not installed for vendor LulzBot, skipping: 
[2023-04-03 14:31:24.198360] [0x00007f81a2b18880] [info]    Confing bundle not installed for vendor BIQU, skipping: 
[2023-04-03 14:31:24.198372] [0x00007f81a2b18880] [info]    Confing bundle not installed for vendor Voron, skipping: 
[2023-04-03 14:31:24.198384] [0x00007f81a2b18880] [info]    Confing bundle not installed for vendor Jubilee, skipping: 
[2023-04-03 14:31:24.198395] [0x00007f81a2b18880] [info]    Confing bundle not installed for vendor Geeetech, skipping: 
[2023-04-03 14:31:24.198409] [0x00007f81a2b18880] [info]    Confing bundle not installed for vendor RatRig, skipping: 
[2023-04-03 14:31:24.202791] [0x00007f81a2b18880] [debug]   Vendor: Templates, version installed: 1.0.1, version cached: 1.0.1
[2023-04-03 14:31:24.202811] [0x00007f81a2b18880] [info]    Confing bundle not installed for vendor Anker, skipping: 
[2023-04-03 14:31:24.202823] [0x00007f81a2b18880] [info]    Confing bundle not installed for vendor Anycubic, skipping: 
[2023-04-03 14:31:24.202835] [0x00007f81a2b18880] [info]    Confing bundle not installed for vendor MakerGear, skipping: 
[2023-04-03 14:31:24.202847] [0x00007f81a2b18880] [info]    Confing bundle not installed for vendor Print4Taste, skipping: 
[2023-04-03 14:31:24.202858] [0x00007f81a2b18880] [info]    Confing bundle not installed for vendor Elegoo, skipping: 
[2023-04-03 14:31:24.202872] [0x00007f81a2b18880] [info]    Confing bundle not installed for vendor INAT, skipping: 
[2023-04-03 14:31:24.202886] [0x00007f81a2b18880] [info]    Confing bundle not installed for vendor Ultimaker, skipping: 
[2023-04-03 14:31:24.202901] [0x00007f81a2b18880] [info]    Confing bundle not installed for vendor Zonestar, skipping: 
[2023-04-03 14:31:24.202915] [0x00007f81a2b18880] [info]    Confing bundle not installed for vendor FLSun, skipping: 
[... cut ...]
[2023-04-03 14:31:24.659562] [0x00007f819bfff640] [info]    Resource PrusaResearch / SL1S_thumbnail.png found in resources folder. No need to download.
</pre>
</details>

Then click the Arrange button:

```
[2023-04-03 14:31:28.684912] [0x00007f81a2b18880] [info]    Undo / Redo snapshot taken: Arrange, Undo / Redo stack memory: 0MB Resident memory: 324MB; Shared memory: 147MB; Private memory: 178MB; Peak memory usage: 756MB
zsh: segmentation fault (core dumped)  prusa-slicer-console --loglevel=9 test.3mf
```

lukasmatena, 2023-04-05:
@monkeydave Thanks, we can now reproduce and fix it.
SPE-1637

closed 2023-05-13 by lukasmatena