prusa3d/PrusaSlicer #10244: Prusa Slicer 2.6.0 a6 - Unable to add physical networked printer

DrJuJu, 2023-04-03:
### Description of the bug

Unable to add a Physical LOCAL networked printer, i.e. a Voron2 driven by a Duet control system, but when trying to add printer I get this error:

Could not connect to FlashAir: HTTP 404: 
Note: FlashAir with firmware 2.00.02 or newer and activated upload function is required.

All previous releases work ok

### Project file & How to reproduce

Try & add Physical Printer

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0-Alpha6

### Operating system

Windows 10

### Printer model

Voron2|Duet

DrJuJu, 2023-04-04:
A bit more info....

I tried connecting to one of my Prusa machines (mK3s), that use Octoprint, and all was as expected, but one thing I did notice was that the "Host Type:" dropbox had an additional 2 entries for "PrusaLink" & "PrusaConnect" which is not available on the other printer types ....

kalpeke, 2023-04-07:
Hi am getting the same issue and I also test using curl and i am able to connect to duet. please see below images
[image]
[image]
[image]

crpalmer, 2023-04-09:
It looks to me like there is mismatch between the text displayed in the "Host Type" dropdown and the internal values.

Trying changing "Duet" to "Mainsail/Fluid" and clicking test.  When I do so it says: "Connection to Duet works correctly".

kocikdav, 2023-04-11:
Hello, there seems to be an error in the Physical Printer dialog. So when you select Host Type, another type is actually used. The names in messages should be correct (while there is different incorrect name in Host Type) . This will be fixed in next release. Meanwhile we there is a test build for windows that should fix this problem here https://github.com/prusa3d/PrusaSlicer/issues/10210#issuecomment-1493976330.

closed 2023-04-11 by kocikdav