prusa3d/PrusaSlicer #10247: PrusaSlicer 2.6.0 Alpha6 Organic supports inside of the object

danielfmontero, 2023-04-03:
### Description of the bug

PrusaSlicer generates part of the organic supports inside of the object without being necessary. 

[image]

### Project file & How to reproduce

[TTTS_Case_PrusaIssue.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/11142476/TTTS_Case_PrusaIssue.3mf.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0-alpha6+MacOSx64

### Operating system

macOS 13.2.1

### Printer model

Original Prusa i3 MK3S+

Firefighter1023, 2023-04-05:
I'm having similar issues, including with other objects on the same platter.

sthompson-celerity, 2023-04-05:
I am on windows same issue. see below

[image]
[x gantry pi case chain.zip](https://github.com/prusa3d/PrusaSlicer/files/11163781/x.gantry.pi.case.chain.zip)

kubispe1, 2023-04-19:
SPE-1663

kubispe1, 2023-05-18:
Collisions of objects and their organic supports should be solved in 2.6.0-beta1. If you encounter any persisting problems, please let us know.

closed 2023-05-18 by kubispe1