prusa3d/PrusaSlicer #10248: External perimeter being printed twice for Classic perimeter generator

rsurgiewicz, 2023-04-03:
### Description of the bug

Hi Folks,

I encountered a problem that in one of my model is (imho) not slicer correctly.

Model is attached (simple box with cutouts in walls). Model was designed to be printed on 0.6 in a meaning that wall width is adjuster to fit 4 perimeters (i guess it's irrelevant).

This is how it looks like (the model)
[image]

For **Classic** perimeter generator
Problematic area where path is actually printed twice:
First time when printing 1st external perimeter:
[image]

Second time when printing innermost external perimeter:

[image]

[image]

GCode clearly indicates 0.7mm material being extruded, that in the end causes final print to look ugly:
[image]

On **Arachne** however only the single small segments are printed separately and makes the final print looks even worse (not relevant) 
[image]

Issue is present on both 2.5.x and 2.6.x-alpha

### Project file & How to reproduce

1. Open Model in PS 2.5.x or 2.6x or open attached project
2. Select Printer with 0.6 nozzle e.g Prusa Mk3 @ 0.6 nozzle
3. Select profile for 0.6 nozzle e.g 0.30mm Quality
4. Select Classic Engine perimeter generator
PRUSA SLICER 2.5 Project [piorniki_projectPS.zip](https://github.com/prusa3d/PrusaSlicer/files/11142498/piorniki_projectPS.zip)
MODEL ONLY: [piornik_model.zip](https://github.com/prusa3d/PrusaSlicer/files/11142501/piornik_model.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.5.2+win64 / Version 2.6.0-alpha3+win64 

### Operating system

Windows/Linux

### Printer model

Prusa Mk3s @0.6 nozzle

rtyr, 2023-04-05:
That is normal behavior with classic perimeter generator, which allows very big perimeter overlap if there is not enough room for standard perimeters.

murk-sy, 2025-02-22:
I was gonna create a duplicate since I've had that issue but for inner perimeters.
I don't really use classic very much but i was testing it trying to resolve extrusion issues with TPU. Ended up resolving it by changing `Print Settings > Advanced > Arachne Perimeter Generator > Minimum perimeter width` to 40%. That made it work a bit more like "Classic" slicing by not going over max extrusion width. This won't help the above issue of interrupted extrusions but might as well post about it.

For future reference:
**Overlapping extrusions with classic perimeter generator**

[image]

[image]

All the following extrusions are two lines of the same extrusion width, with only the very topmost layer being a single line of gap fill.
(Side view)
Enabling or disabling gap fill doesn't change the behaviour.

[image]

I guess classic can't really do multi line gap fills or non-polygon inner perimeters.

kubispe1, 2025-05-07:
Regarding the current behavior of the Classic perimeter generator, I am adding one more model from a customer that exhibits the same issue. It is a narrow wall object where two extrusions do not fit next to each other, resulting in what appears to be two passes of the same extrusion.
[Gardena sprinkler filter v5 (1).zip](https://github.com/user-attachments/files/20076939/Gardena.sprinkler.filter.v5.1.zip)

[image]

closed 2026-09-29 by prusaslicer-bot[bot]