prusa3d/PrusaSlicer #10249: [FR] Externalize printer/slicing/material profiles.

cmdremily, 2023-04-03:
**Is your feature request related to a problem? Please describe.**

This FR would solve two problems for me:

1) I perform slicing for my my printer on several machines depending on what's most convenient for me. I would like to be able to access an up to date copy of my printer, slicer and material profiles on both machines and be able to adjust them on either machine without manual juggling of files back and forth.

2) I would like to have automated backups of my profiles that I've spent a lot of time on tuning them so that I don't lose them during a drive crash (which happened recently 😭).

**Describe the solution you'd like**
Please allow me to select a folder or file on a local computer where the above profiles are stored and read from instead of the default location.

I would then on each of my computers select a folder/file that is synched to DropBox/Google Drive/OneDrive or similar.

This means that my profiles are automatically backed up and synched between devices that I care about.

**Describe how it would work**
This would entail storing the profiles in a configurable path. The path would be configured in the GUI and stored in a settings file that is in a pre-determined path as it is now.

Some method of preventing a data race that could corrupt the profile file is necessary.  A common solution is [lock-files](https://en.wikipedia.org/wiki/File_locking) that would probably need to be implemented using files as I don't know of a portable cross platform solution.

These externalized profiles should also either contain the system profiles that then get automatically updated. Or contain a list of system profiles that are installed so that they appear on both machines and I don't need to re-do initial setup on both and wonder which material profiles I selected when installing the previous machine, or update when I add in one of the system profiles.

**Describe alternatives you've considered**

Making Prusa Slicer Cloud Connected: I do not want this. I don't want my slicer to be "cloud enabled". I don't think it's necessary, I think it will be too much effort to build up the infrastructure for this and it incurrs cost. It'll add a dependency on the cloud to load my proiles, unless additional work is done with caching, but that has corner cases such as first set-up. It'll open up data privacy questions etc. I think this is a bad option.

Integrating directly with cloud storage APIs: This seems less flexible than just using local files and letting the user set up the synching as they please. It imposes additional testing and maintenance overhead on the developers, while by necessity making some smaller/lesser known cloud storage solutions unsupported. It would probably be more convenient for users who are not that tech savvy, but I think that that's a smaller cohort of makers in general. Maybe this can be offered as a nice to have in addition to the proposed solution if there's a desire for it. 

Doing nothing: This is always an option, and I'll have to struggle and be annoyed at manual file copying in 2023 :)

**Additional context**
N/A

Area5142, 2023-04-03:
PrusaSlicer has the command line option --datadir that defines where the configuration files are located.
Here the description from prusaslicer when executing with --help option:

`--datadir ABCD      Load and store settings at the given directory. This is useful for maintaining different profiles or including configurations from a network storage.
`

I have used this setting with network drives to have "one" configuration for different prusaslicer installations that had to have the same configuration files.

Not sure of this solves all your requests.

foreachthing, 2023-04-04:
I have use that in past, too.
But came away from it because when you double click a 3mf file, it will open PS _without_ the settings. I did not spend time to figure out the batch file, which can then be associated with 3mfs that opens PS and loads its settings.
Should be fairly easy though. But I am the only one printing, so ... 

But, yeah! I wish there was a setting in the GUI, or in the .ini at least, so this association-mess with filetypes would not be affected.

my 2¢:
_If I had a multi-user/-machine env, I probably would distribute the base profile and then the user can adjust/tune/change the settings further. That way, one "screwed-up" setting does not propagate to every user/machine. Profiles are only reloaded when PS restarts and everybody can save their settings at any time, so you don't know what settings you end up with next time PS starts.
If those settings need to be on a server/nas, every user had a dedicated share to store such information on. As admin it would be easy to sync these settings manually or via cron - which I would not recommend._

murk-sy, 2023-04-21:
+1, I use datadir and was just preparing a bug report on "Start new Prusaslicer instance" not using --datadir. I keep the binaries and the config on a network drive so I can access it from other machines as well if needed - plus it's redundant there, unlike locally.