prusa3d/PrusaSlicer #10251: Prusa Slicer update try to downgrade 2.5.2 to 2.5.0?

NightSkySK, 2023-04-04:
### Description of the bug

From the prusa3d.com I've installed Drivers&Apps package 2.5.2 (as I moved myself to fresh computer) after launching the Prusa Slicer automatic update try to downgrade Prusa Slicer to 2.5.0 
[image]

Is it expected behavior?

### Project file & How to reproduce

[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.2 +win64

### Operating system

Windows 10

### Printer model

Prusa i3 Mk3

kubispe1, 2023-04-04:
@Janys, please can you check it? Thanks

Janys, 2023-04-04:
Hello, we had an old update file cached on the server. It should be fixed now, apologies for the inconvenience!

NightSkySK, 2023-04-04:
Thank you for the fast reaction and the fix implementation!

NightSkySK, 2023-05-09:
Sorry to reopen the issue but I found that issue is not fixed entirely. 
[image]

Prusa Sliser still try to update it self. This time it tries to install the same version which is already installed. (2.5.2)

closed 2026-09-29 by prusaslicer-bot[bot]