prusa3d/PrusaSlicer #10252: tree supports

xanaddixt, 2023-04-04:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe how it would work**
Be as descriptive as your skill level allows you.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or paint drawing about the feature request here.

xanaddixt, 2023-04-04:
No my issue isn't fixed, how do you get tree supports and control them on
prusa because I'm not seeing any settings for tree supports available on
advanced settings?

jmd73, 2023-04-04:
@xanaddixt First you need to use one of the alpha versions of PrusaSlicer 2.6.0. Can be downloaded from Github.
Then you can choose organic as support style.
[image]