prusa3d/PrusaSlicer #10256: Bed temperature for other layers not respecting 0 setting!

adolson, 2023-04-04:
### Description of the bug

I have set the bed temperature to 0, and the tooltip says "Set this to zero to disable bed temperature commands in the output" however, it is not being respected. I clearly see this in my g-code after the first layer:

M140 S0 ; set bed temperature

Furthermore, it does not appear to respect when the printer setting "Emit temperature commands automatically" is unticked. Perhaps this setting is ONLY for the start g-code, but I would have expected it to disable all temperature commands (including the one that already should have been disabled).

### Project file & How to reproduce

[Hex_Wrench_Handle_with_Screw.zip](https://github.com/prusa3d/PrusaSlicer/files/11151007/Hex_Wrench_Handle_with_Screw.zip)

Generate g-code, check the code, find the M140 S0 command in the code.

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0 alpha 6

### Operating system

Debian Bullseye

### Printer model

Eryone Thinker SE (heavily modded)

neophyl, 2023-04-04:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/8410

adolson, 2023-04-04:
> Duplicate of #8410

My bad. I did search but didn't go that far back, as I thought it might've been related to the new Klipper changes.

Still, this should be fixed.

closed 2023-04-04 by adolson