prusa3d/PrusaSlicer #10259: Incorrect extrusion width when slicing

rainell, 2023-04-05:
### Description of the bug

I explain step by step it in the next field.
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

### Project file & How to reproduce

1- Add any shape;
2- click Slice now;
3- switch the preview mode to Width and look at the result (image 1), everything will be printed with the lowest extrusion width of the graph, which is equal (or approximately equal) to the Top solid fill (image 2).
4- In the Star G-code, comment out the extruder movement during the purge line (image 3); 
5- slice again, now each part of the print has the correct extrusion width (image 4).
6- You can perform the same test with another printer, for example an Artillery Sidewinder X1 (images 5, 6, 7 and 8).

PS: I have not verified the real wall width of a piece printed in vase mode because I use Klipper, that is to say I don't use the default Start G-code; So I don't know if it's just a visualization problem or also a slicing problem.

[Shape-Box-2.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/11154207/Shape-Box-2.3mf.zip)
[Shape-Box.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/11154096/Shape-Box.3mf.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0-alpha6+MacOS-x64 / 2.5.1+MacOS-x64 / 2.5.1+win64 / -2.6.0-alpha6+win64

### Operating system

macOS 10.15.7 / Windows 10

### Printer model

Any (Prusa, Artillery, Ender, etc.)

neophyl, 2023-04-05:
I think its just a preview scale issue.  Your prime lines are 'fat', hence the scale going to 0-2mm extrusion width.  At that scale on the legend everything thats printed with normal extrusion widths looks almost the same colour and its impossible to differentiate the difference between 0.4 and 0.45.  The actual printed gcode should be fine.  

I'm sure this has been reported before.  Ah found one of them https://github.com/prusa3d/PrusaSlicer/issues/8479

kubispe1, 2023-04-05:
and also #8777

rainell, 2023-04-09:
@neophyl in any case, it is a problem for users, no matter how you justify it from your point of view as an advanced or expert user. When users go to the extrusion width preview they see wrong information.

FidelCapo, 2023-04-24:
Issue should be fixed by adding new button in legend and fix will be part of next release.
Closing.

closed 2023-04-24 by FidelCapo