prusa3d/PrusaSlicer #10262: morceaux tombe sur le lit

18pool49, 2023-04-05:
### Description of the bug

quand je charge un groupe de STL ( 7 p.)  
je me retrouve avec ma pièce en 1 seul morceau bien empilé que je suis obligé de "scinder" pour retrouver mes 7 morceaux 

mais voilà que les morceaux en hauteur qui ont pas de contact avec le lit >>>> sont PLAQUÉS sur le lit au pied des autres morceaux , C'est un puzzle avec aucune référence de position 
 
[image]
 ca c'est la pièce de 7 morceaux FINIE 

[image]     grrrrr  ca c'est ce que je retrouve une fois >> "SCINDER" 

ma question est ?????    comment remettre les pièces a leur place >>> ou comment éviter ce PUZZLE   GRRRRRR! ! ! 

### Project file & How to reproduce

[cr 7p. id-7parts-export rapport prusa.zip](https://github.com/prusa3d/PrusaSlicer/files/11158258/cr.7p.id-7parts-export.rapport.prusa.zip)
il y a 3 morceaux qui sont des bavures de la découpe que Prusa a trouvé  et qui sont inexistant avec Ideamaker qui les a tranché 
??????? pourquoi ???????  

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.0.6. alp.6

### Operating system

W10

### Printer model

A10M geeetech

neophyl, 2023-04-05:
Use split to Parts not split to Objects.  
Each Object must have contact with the bed.  An Object only has to have a single PART in contact with the bed, the rest of the Parts may be placed however you like.  When splitting to Parts they stay in the same relative positions.

18pool49, 2023-04-05:
merci pour m'avoir éclairé ,mon incompétence , ca marche beaucoup mieux 
merci beaucoup
[image]
[image]