prusa3d/PrusaSlicer #10264: Fuzzy skin modifiers that do not alter the perimeters

csweeney0926, 2023-04-05:
I am using the new prusa slicer 2.6.0 alpha and am running into a problem with the fuzzy skin feature. 

Context:
I like to use fuzzy skin on final products as it provides a textured surface and can effectively hide layer lines, making it less obvious that the object in question is 3d printed. 

I have been working on a mechanical part that has interlocking components that have tight tolerances as well as large areas that are not dimensionally critical that could be improved by using fuzzy skin. I attempted to have the best of both worlds by applying fuzzy skin to the outside walls and using modifiers to remove fuzzy skin from the dimensionally critical sections of the object (like gear teeth). 

The problem I am running into is the volume modifiers appear to enforce a new set of perimeters at the edge of the modifier. This is a problem because the gear teeth need to be structurally sound as well as having accurate dimensions. Having a perimeter on the edge of the modifier effectively makes the gear teeth very weak as it introduces a seam at the edge of the modifier.  

Normally I would just separate the object into individual components (and just apply fuzzy skin to the relevant pieces) but unfortunately this object is print-in-place so it all has to be one component.

Would it be possible to have a volume modifier option that does not introduce new perimeters at the edges of the volume modifier?

foreachthing, 2023-04-12:
I wish!
So far I've been using Blender to duplicate the faces I like to make fuzzy. Then apply a solidify modifier (in Blender - just make it just thick enough not to disturb all perimeters) and use that mesh as modifier in Prusa Slicer. Bit of a workflow but once you get to know Blender, not as bad.
But that "thing" with modifiers in PS goes back very far.... and still no progress or any light of this long and dark tunnel.

csweeney0926, 2023-04-13:
> I got a height range modifier to work with fuzzy skin, and print threaded containers like this (with fuzzy skin enabled only until height z=...). This of course only works if the unfuzzy parts are in a certain z range, so a super narrow set of cases. I would like to see your feature request to be implemented as well!

csweeney0926, 2023-04-13:
Yeah I am aware of the z-height g-code modifier approach but that won't work for this model either (too many parts at different elevations). I don't use blender and that approach sounds like it would get very complicated for this model as well.

The apply to face is an interesting approach though. 2.6 alpha already has some capability to identify faces of a model in the feature that allows to to select a face to place on the bed. I wonder if that programming could be modified to allow for applying fuzzy skin to selected faces? 

Another approach would be to do something like the support modifier on painted areas. If you could pain the areas where you did (or alternatively did not) want fuzzy skin to be applied that would be a good approach as well

foreachthing, 2023-04-13:
Paint-on fuzzy skin: https://github.com/prusa3d/PrusaSlicer/issues/6988

foreachthing, 2023-04-13:
> I don't use blender and that approach sounds like it would get very complicated for this model as well.

Challenge accepted ;-)

lukasmatena, 2024-11-21:
Fixed in [2.9.0-alpha1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.9.0-alpha1). Closing.

closed 2024-11-21 by lukasmatena