prusa3d/PrusaSlicer #10265: Prusa Slicer 2.6.0 Alpha 6 - Problem with exporting STL with cut connectors

csterlin, 2023-04-05:
### Description of the bug

Better, but still buggy: On some objects cuts with connector dowels works as expected. On others it does not, displaying an error message: Unable to perform boolean operations on model meshes. Only positive parts will be exported.
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

### Project file & How to reproduce

[HULK_003.zip](https://github.com/prusa3d/PrusaSlicer/files/11163890/HULK_003.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.6.0-alpha6+win64

### Operating system

Edition	Windows 11 Home Version	22H2 Installed on	‎10/‎1/‎2022 OS build	22621.1413 Experience	Windows Feature Experience Pack 1000.22639.1000.0

### Printer model

Creality Ender 3 (original with 4.2.7 card with firmware from Creality: Endr-3Marlin2.0.6HW4.2.7

dehoresh, 2023-04-16:
I also have this issue when trying to export objects as individual STL files.

wwian, 2023-05-10:
I also have this issue when trying to export objects as individual STL files.

rdbischoff, 2023-07-05:
I kind of took the error as their way of trying to make you use their slicer to do the slicing. 🤷‍♂️ if that’s not true then awesome. Cuz I’m so much more used to Cura for slicing. I’d like to learn Prusaslicer but it looks confusing to me.

bradcarnage, 2024-01-11:
+1 on prusa slicer 2.7.1

marcosy87, 2024-01-16:
I've found that the problem can be solved by right clicking on the piece>"simplify model" and then "fix by windows repair" (prusa slicer 2.7.1, this didn't work in 2.6.0 alpha 6)

RcrossF, 2024-02-19:
This is still an issue, Simplifying and repairing had no effect for me. Funny enough it'll slice no problem, but I can't export it as STL.

[image]

Jugernaunt, 2024-04-08:
I keep trying to resize and move the dowels and even went from 2-1. I tried to fix by windows repair and still wont work. I know people are doing it in Blender but many more steps. Couldn't there be a work flow to do the same process all at once. We got AI now and this cant happen?

closed 2026-09-29 by prusaslicer-bot[bot]