prusa3d/PrusaSlicer #10268: PrusaSlicer 2.6.0 Alpha 6 - Purging Volumes for multi extruder/single hotend setup

joshua70448, 2023-04-06:
### Description of the bug

I've modified my Ender 3 v2 to run 4 extruders and combine them with a 4-in-1-out adapter before going to a single hotend. This has worked well so far, but with the latest Alpha 6 release, the Purging Volumes button was removed. The release notes say "Setting purging volumes based on which filament is unloaded does not make sense for proper multi-extruder systems, which can easily control the required purging in Filament Settings->Advanced->Minimal purge on the wipe tower." but this doesn't work well for me. I'd much rather run one filament preset and be able to specify the purging volumes per color the way I could before, rather than having to defined presets for each individual color and plug in a constant purge volume. This prevents me from optimizing the purge amount by color transition (for example, going from yellow to white doesn't require as much purging as black to white). Could you please bring the Purging Volumes button back for setups like mine?

### Project file & How to reproduce

[Quad Cube.zip](https://github.com/prusa3d/PrusaSlicer/files/11166935/Quad.Cube.zip)
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0-alpha6-win64

### Operating system

Windows 10

### Printer model

Ender 3 v2 with Octopus board and 4 extruders

lukasmatena, 2023-04-06:
Why don't you just check "single extruder multi material" in Printer Settings? The description of your setup implies that you should ("4-in-1-out adapter, single hotend").

joshua70448, 2023-04-06:
I'll give it a shot, although I had to zero out a bunch of settings since I already handle ramming and parking and unparking in my tool change macro codes, not certain if there's anything else that might act unexpectedly.

closed 2023-04-11 by joshua70448