prusa3d/PrusaSlicer #10269: Dynamic Overhang Speed issues

rednaxal, 2023-04-06:
### Description of the bug

Dynamic Overhang Speed is having couple issues.
1. It says that in case a percentage is used then it calculates it from external perimeter speed. It seems like it is taking not the external perimeter speed only but the actual perimeter speed. For example in case of a small perimeter it takes the small perimeters speed as a base. So using percents is basically results in undefined behavior. 
2. If '0' is left for any of the settings then this results in 0 speed for an overhang perimeter:
[image]
In printing this results in nearly zero speed (on Klipper) on such places which results in everything melted around a nozzle, because it just stops moving for a few seconds.  
It either should not accept 0 or warn or ignore the setting in case it is set to 0 I believe. 

### Project file & How to reproduce

[Bug.zip](https://github.com/prusa3d/PrusaSlicer/files/11167691/Bug.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0-alpha6+win64

### Operating system

Windows 10

### Printer model

Kingroon KP3S

yodor, 2023-04-09:
I hit this issue also, it says it will interpolate the speed but then it is not doing so. I have G1 F0 in gcode and then printer halts for incorrect speed 0.

Jan-Soustruznik, 2023-04-12:
SPE-1650

rednaxal, 2023-04-18:
In addition to the problem above I've discovered another problem with this new feature. For some reason it interpolates speeds wrongly even when it has been set explicitly in mm per sec and not in percents. Example:
[image]
Why is it setting 4.1mm/s when it is supposed to be not less than 10mm/s:
[image]
Is it somehow considered this overhang to have a negative overlap? :)

FidelCapo, 2023-04-19:
This issue should be fixed and fix will be part of next release.
Closing.

closed 2023-04-19 by FidelCapo