prusa3d/PrusaSlicer #10277: Spelling error ?

beadon, 2023-04-07:
### Description of the bug

Main window has a spelling error.

"Plater" is not an English word that makes sense in this context:
https://www.merriam-webster.com/dictionary/plater

I think you mean: "Platter" : https://www.merriam-webster.com/dictionary/platter

However, many printers use the english term "bed" : 
see definitions 2,3,5 : https://www.merriam-webster.com/dictionary/bed

### Project file & How to reproduce

[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.2

### Operating system

macOS 13.3

### Printer model

MakerBot Replicator Gen 5

neophyl, 2023-04-07:
https://github.com/prusa3d/PrusaSlicer/wiki/Frequently-Asked-Questions-(FAQ) First entry.

J3r0github, 2023-04-10:
For future refrence, it's always good to check that your issue is not a duplicate.

beadon, 2023-04-29:
> https://github.com/prusa3d/PrusaSlicer/wiki/Frequently-Asked-Questions-(FAQ) First entry.

Ah.  Well, seems like there is quite the debate.  I will close this issue.  Thank you for the polite reply.

beadon, 2023-04-29:
> For future refrence, it's always good to check that your issue is not a duplicate.

Very fair.  I didn't find it on a quick search, but should have looked a bit harder.  Issue closed.  Thank you.

closed 2023-04-29 by beadon