prusa3d/PrusaSlicer #10278: App crash when pressing auto arrange....

PaulN8, 2023-04-07:
### Description of the bug

Application hung (not responding) then crashed after pressing the auto arrange button with a single .3MF imported object that extends beyond the bounds of the build plate. Tested repeatedly and crashes consistently, even with just a simple cuboid object.

### Project file & How to reproduce

[Auto arrange crash.zip](https://github.com/prusa3d/PrusaSlicer/files/11176573/Auto.arrange.crash.zip)
[image]

Import object the extends beyond build plate dimensions.
Press the auto-arrange button.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0-alpha6+win64-202303311519

### Operating system

Windows 10

### Printer model

MINI+

sroemer, 2023-04-10:
I can confirm this bug. I am also running the latest alpha 6 build of 2.6.0, but in my case on Linux x86_64 and with an Elegoo Neptune 3 Pro. My object also sticks over the edges of the build plate. When resizing first and then arrange, the crash does not happen.

Jan-Soustruznik, 2023-04-11:
SPE-1652

tamasmeszaros, 2023-04-17:
Same underlying problem as in #10241

closed 2023-05-13 by lukasmatena