prusa3d/PrusaSlicer #10279: Slicing changing object

SorenWHartmann, 2023-04-07:
### Description of the bug

When slicing an object, the form object is changed.
The two flanges on the side changes shape

### Project file & How to reproduce

[image]
[image]
[GC_BOP STACK 11-15k in parts with Logo v1.zip](https://github.com/prusa3d/PrusaSlicer/files/11176830/GC_BOP.STACK.11-15k.in.parts.with.Logo.v1.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.2 + Win64

### Operating system

Window 11

### Printer model

GC_BOP STACK 11-15k in parts with Logo v1

SorenWHartmann, 2023-04-07:
Have a nice Easter :)

neophyl, 2023-04-07:
Your model has issues. See the warning triangle against it in your first picture in the tree view. Fix the model.

Area5142, 2023-04-07:
The Step model has errors that needs to be fixed before importing into PrusaSlicer.
[image]

Here is the fixed model converted to STL for import to PrusaSlicer:
[GC_BOP STACK 11-15k in parts with Logo v1_Fixed.zip](https://github.com/prusa3d/PrusaSlicer/files/11179503/GC_BOP.STACK.11-15k.in.parts.with.Logo.v1_Fixed.zip)

closed 2023-04-11 by Jan-Soustruznik