prusa3d/PrusaSlicer #10280: Extrusion Multiplier for each Feature Type

arshish1612, 2023-04-07:
At the moment we just have a single extrusion multiplier. Can we get this setting for each feature type? This is a setting available in Cura abut it is missing in Prusa Slicer
[image]

Tupson444, 2023-04-07:
This would be very useful for high flow printing, for example, if infill is printed with higher volumetric flow (which means slight underextrusion) compared to perimeters. My current workaround is using Gcode substitutions:

>replace:
>;TYPE:Perimeter
>
>with:
>;TYPE:Perimeter\nM221 S100 ; set flow to 100%

and

>replace:
>;TYPE:Solid infill
>
>with:
>;TYPE:Solid infill\nM221 S110 ; set flow to 110%

etc.

But it would be easier if there were settings for this (or something like "Dynamic flow compensation" as suggested here: #10168 )

rtyr, 2023-04-07:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/7200.

closed 2023-04-07 by rtyr

arshish1612, 2023-07-17:
When can we get this feature? Its an absolute Must have for clean top layers. Even OrcaSlicer has it implemented for Top layer.

Henk72, 2023-12-17:
You can use a post-processing script that does the substitution for all layers except the first. Search for a good script and change it to your needs.