prusa3d/PrusaSlicer #10281: When Background Processing is enabled, any ongoing manipulation, like move, rotate gets cancelled when the processing completes.

viesturz, 2023-04-07:
### Description of the bug

When auto slicing is enabled, any ongoing manipulation, like move, rotate gets cancelled when the slicing completes.

I expect editing can be done normally, regardless of auto-slicing setting. With this the auto slicing mode is very hard to use.

### Project file & How to reproduce

1. Configuration -> Preferences -> General -> Background Processing -> Enable
2. Load a large-ish STL file that takes several seconds to process
3. Start moving the object around using mouse drag.
4. Keep dragging, do not let the mouse button go, wait until the background processing complete.

Video: https://photos.app.goo.gl/RwsQ1BvESwZpzAhJ8

Expected: 
- you can keep moving the model around uninterrupted
- the move operation is cancelled, the object jumps back to it's previous position.

My recommendation would be to proactively cancel the current ongoing processing if user starts a manipulation, the processing result will be discarded anyway.

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0-alpha6

### Operating system

Win11

### Printer model

voron 0.2

veikman, 2024-07-27:
Symptoms are identical to #6898.

I am still seeing this problem in 2.8.0+linux-x64-GTK3.

tuxmobil, 2024-08-18:
This behaviour is annoying me as well. I agree with the suggestion that the background processing should be cancelled once an object is being moved.

closed 2026-09-29 by prusaslicer-bot[bot]