prusa3d/PrusaSlicer #10282: Wrong auto extrusion width

Sprutx, 2023-04-07:
### Description of the bug

If I manually set the extrusion width, then everything is fine

If I set it to 0, then the description says that the nozzle width will be used, but a different value is used

G-code 1:

; nozzle_diameter = 0.4
; extrusion_width = 0

; external_perimeter_extrusion_width = 0
; first_layer_extrusion_width = 0
; infill_extrusion_width = 0
; perimeter_extrusion_width = 0
; solid_infill_extrusion_width = 0
; support_material_extrusion_width = 0
; top_infill_extrusion_width = 0

; external perimeters extrusion width = 0.45mm
; perimeters extrusion width = 0.45mm
; infill extrusion width = 0.45mm
; solid infill extrusion width = 0.45mm
; top infill extrusion width = 0.40mm
; support material extrusion width = 0.40mm

G-code 2:

; nozzle_diameter = 0.4
; extrusion_width = 0.4

; external_perimeter_extrusion_width = 0
; first_layer_extrusion_width = 0
; infill_extrusion_width = 0
; perimeter_extrusion_width = 0
; solid_infill_extrusion_width = 0
; support_material_extrusion_width = 0
; top_infill_extrusion_width = 0

; external perimeters extrusion width = 0.40mm
; perimeters extrusion width = 0.40mm
; infill extrusion width = 0.40mm
; solid infill extrusion width = 0.40mm
; top infill extrusion width = 0.40mm
; support material extrusion width = 0.40mm

### Project file & How to reproduce

[test.zip](https://github.com/prusa3d/PrusaSlicer/files/11178681/test.zip)

extrusion width set to 0

check extrusion width in G-code

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0-alpha6 win64

### Operating system

Windows 10

### Printer model

KingRoon KP3S Pro

rtyr, 2023-04-07:
> If I set it to 0, then the description says that the nozzle width will be used.

The description does not say that.

See the screenshot below.
[image]

I am closing the issue.

closed 2023-04-07 by rtyr

Sprutx, 2023-04-10:
So this is not the correct translation into Russian, it says - "if 0, then the nozzle diameter will be used"