prusa3d/PrusaSlicer #10283: Internal USB card reader is broken until reboot by new Eject functionality in 2.6.0-alpha6-Win64

jfinnie, 2023-04-07:
### Description of the bug

Using the eject functionality breaks my PC case's internal USB card reader until next reboot.  
USB card reader is:
Alcor Micro USB 2.0 Card Reader
USB\VID_058F&PID_6366&REV_0100

Windows says this to me:
"Windows cannot use this hardware device because it has been prepared for "safe removal" but it has not been removed from the computer. (Code 47)

To fix this problem, unplug this device from your computer and then plug it in again."

Of course, I can't unplug it, as it is internal!  So the only option is a reboot.

This USB card reader is bundled with the Coolermaster Silencio S400 case.  Works flawlessly in 2.5.1.

### Project file & How to reproduce

I don't think config related, but see attached empty config:
[empty.zip](https://github.com/prusa3d/PrusaSlicer/files/11178833/empty.zip)
Screenshot of the effect of the Eject:
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0-alpha6+win64

### Operating system

Windows 10 Version 10.0.19044 Build 19044

### Printer model

Prusa MK3S

kocikdav, 2023-04-11:
Hello, thank you for reporting this. We are actively working on the fix. There is a issue #9881 with problems connected to ejecting in 2.6.0. But nobody reported problems with internal card reader so far so I will leave this issue open.

jfinnie, 2023-04-17:
This issue is fixed by the build linked in:
https://github.com/prusa3d/PrusaSlicer/issues/9881#issuecomment-1511384451

closed 2023-04-18 by jfinnie