prusa3d/PrusaSlicer #10284: PrusaSlicer 2.6 Rotate connectors in Cut Tool

YoloDoge, 2023-04-07:
**Is your feature request related to a problem? Please describe.**
No problem, I'm just annoyed.

**Describe the solution you'd like**
In case of choosing a connector shape of square, rectangle or hexagon, I want to be able to rotate the shape, so that one of the sides of the shape can be parrallel to the bed. In that way the support can do it's optimal job by supporting the protrusive part of the connector.

**Describe how it would work**
Within the 'Add connectors' submenu of the Cut Tool, add a box section 'Rotation' where you can put in the angle degree of the rotation. Most optimal would be to select a face of the connector shape to auto snap parrallel to the print bed.

**Describe alternatives you've considered**
As far I have seen, there are no alternatives. After the cut, I selected the 'Cut connectors' feature in the feature tree, in the hope that the rotation option on the left pane became available, but to no avail.

**Additional context**
See attached picture
[image]

Artemonim, 2023-09-04:
This is especially important since snap connections have become available:
[image] [image]

YuSanka, 2023-09-05:
Thanks for your report. (SPE-1860)

Naeaeaeaeae, 2023-09-15:
Not sure if it is the same or a bit more.
I opened https://github.com/prusa3d/PrusaSlicer/issues/11309 

-> set all dimensions, positions and orientation of Connectors

nrnrnr, 2023-10-07:
Yes!  Crucial for snap connectors.

nrnrnr, 2023-10-07:
Followup:  If the snap connectors are oriented incorrectly, the orientation can be corrected by a series of 90-degree rotations of the cut plane.  Not intuitive, but doable.

lukasmatena, 2023-10-21:
Implemented in [PrusaSlicer 2.7.0-alpha1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.0-alpha1). Closing.

closed 2023-10-21 by lukasmatena