prusa3d/PrusaSlicer #10286: Bridges are printed in the Wrong Direction !!!

Admiral-dk, 2023-04-07:
### Description of the bug

When printing a Bridge -  all the previous versions (Slicer 2.1.0-2.6.0alpha4) Prints Across the 'supports' (OK) - here Slicer 2.5.0 :

[image]

BUT Slicer 2.6.0 alpha6  Prints parallel to the 'supports' (bad term, as it is in use elsewhere) => This results in an unsupported Print maybe 130mm long - instead of 2.6mm Across the channel it should cover ...!!!   :-(
Also note that the Slicer reports an error (Warning) in my design - but it is a result of the error in the Slicer.

[image]

I have attached the *.stp file (see .zip) - so you could try yourself .... 

Win10. No special commands etc. .
config.ini is inside the .zip file
[DL650_COS-E50_r150.zip](https://github.com/prusa3d/PrusaSlicer/files/11180967/DL650_COS-E50_r150.zip)

The design is a small part, inside a bigger design (so it is not floating as you see it here) - but no point in uploading it all.

Best wishes

Per

### Project file & How to reproduce

Just slice the *.stp file and see the same result as in the screenshots @ the same level, where the channel should have a 'Roof on top'. Here you will see the Slicers 2.1.0 - 2.6.0alpha4 all 'Print' the 'Roof/Bridge' @ an angle to the 'Supports/Walls' - this way easily covering it without any problems :-)

BUT Slicers 2.6.0alpha6 tells the Printer to print the whole length (almost) in 'Free Air' - parallel the the 'Supports/Walls' ...!!!

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0-alpha6+win64

### Operating system

Win10 Pro DK 22H2  build 19045.2728

### Printer model

Prusa i3 MK3S+

kieraneglin, 2023-04-09:
Does the issue resolve if you go to print settings > infill > advanced > bridging angle and change that to 90º?

Admiral-dk, 2023-04-10:
Hi Kieraneglin

Thank you for your reply. The short answer - yes it does make a difference .... But also a new problem for a bigger Bridge on another level, where the Slicer had found the optimal Direction, and that will be changed too ....

Now that was Interesting ...!!!
I have always wondered why some of the Bridges in my builds where done the way they where. Now I see that the Zero degrees are Horizontal on the Screen - sometimes no matter the direction of what needed to be Bridged ....
So setting it to 90 degrees as you sugested, makes then Vertical on the Screen ....
As what needs to be Bridged here runs close to 45 degrees to the Screen - you've guessed it - setting it to 45 gives a 90 degree Bridge.
Now this gives me another problem - as I have more than one layer (@ another heights) that needs to be Bridged - but not nessesarry in the same direction ....
Many times it is possible to find a good compromize - BUT - what about all those that arent good with Nerding in Parameters .... They where better served with how the Slicer worked before.

Jan-Soustruznik, 2023-04-11:
Hello, @Admiral-dk , 

Thank you for your report. We deeply checked your issue and found that the new behavior isn't issue. It is pretty funny, but problem was located in 2.5.0 NOT in 2.6.0. In some cases the algorithm generated bridge in "correct" way, but it was mostly accidental. So now the algorithm should work the same for all models according to decided rules - If the bridged area is not surrounded by anchors, then such direction that contains least amount of bridge tips floating in the air is chosen. We do not currently consider touching a perimeter from a side as a sufficient anchor for external bridges, due to aesthetic issues.

In your case(shared model) there isn't space to generate an anchor for bridge > walls are thin, if you change perimeters value to 1 - bridge will be generated same as 2.5.0. 

As as is mention above you can modify the bridging angle, 
1. generally  _print settings >infill > advanced > bridging angle_  
2. using by modifier - you can edit the bridging angle locally. (more: https://help.prusa3d.com/article/modifiers_1767)

I closed issue as solved. 

Thank you, have a nice day

Admiral-dk, 2023-04-11:
Hello Jan

Thank you for your reply.
The Modifier solution works (though quite annoying when I can't see what I'm doing, as this is inside a much bigger container) - the result is only visible after slicing and looking @ the individual layer .... So if not correct - start all over again.
This I can solve by making sure that I only modify one layer and make sure no other Bridges are done @ that Level ...!

You are not correct about the different Slicer versions - I had no problems with the same design in Slicer 2.1.0 2.4.0 2.4.2 2.5.0 2.6.0alpha4 2.6.0alpha5 but only in in 2.6.0alpha6  - oh well ....

I have just tried making the sides 1.0mm (and also 1.2mm) instead of the 0.8mm, as you suggested - but this doesn't make any difference - neither as a stand alone 'Part' as I send you or inside the complete structure ...!

Best wishes

Per

ps   I do understand if it stays closed (even if I kind hope ...)