prusa3d/PrusaSlicer #10288: Slicing fails at low layer widths

JuliaF1350, 2023-04-08:
### Description of the bug

### Description of Bug
When slicing files at low line widths (<0.23mm) and greater than one top layer, prusaslicer returns the following error:

> Flow::spacing() produced negative spacing. Did you set some extrusion width too small?

### Project file & How to reproduce

Error Box:
[image]

File to Reproduce error:
[Shape-Box.zip](https://github.com/prusa3d/PrusaSlicer/files/11182587/Shape-Box.zip)

To reproduce, slice a file with a line width of <0.23mm and >1 top layer

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0-alpha6+win64

### Operating system

win11

### Printer model

Ender 3 v2

FreeWilly2022, 2023-04-11:
I am getting same error.  Model slices with no errors in version 2.5.1.  Tried changing Arachne perimeter generator Minimum feature size to 50% but no effect.

stepikovo, 2023-04-25:
Hello @JuliaF1350, I see you have set the _Default extrusion width_ to 0.18 mm and the rest of the parameteres to 0, with 0.2 nozzle, that seems to be the issue.

Ivan-753, 2023-05-02:
Hi, same error.
v2.6.0-alpha6 (and no error for v2.5.0 / v2.5.2)
0.2 nozzle

have error if: 
infill_extrusion_width = 0.2
solid_infill_extrusion_width = 0.2
(same for 0.21)

model sliced if:
infill_extrusion_width = 0.22
solid_infill_extrusion_width = 0.22

also have:
external_perimeter_extrusion_width = 0.15
support_material_extrusion_width = 0.18
all over extrusion width is 0.2

GregAtPospipe, 2023-06-21:
Version 2.5.2

Same issue with very standard settings that come with Prusa Slicer.  Prusa MK3s with MMU2s.  4mm nozzle

Canned printer profile "Original Prusa i3 MK3S & MK3S+ MMU2S Single".  I haven't modified any of the settings

When I slice the attached STL with the default "0.30mm Draft" setting, I get this error.  But if I change to "0.2mm Speed", I have no issues.

[BrentExtruderCylinder80 v3.zip](https://github.com/prusa3d/PrusaSlicer/files/11795477/BrentExtruderCylinder80.v3.zip)

hejllukas, 2025-01-04:
Not reproducible in 2.9.0 with attached files. Closing.

closed 2025-01-04 by hejllukas