prusa3d/PrusaSlicer #10291: Bottom layer affect any upper surfaces, Ph.BICHET

montagnon25, 2023-04-08:
Good morning Sir,

When i don't get any bottom solid layer, at my lower surfaces, it affects all surfaces, almost they are at 55 mm height .
If i'll get 2 or 3 solid layer at my first bottom layer, my small surface at 55 mm height is good . I think it's strongly diificult to understand, because i don't want any bottom layers, but i want this 55 mm height little surface.
[image]
[image]

neophyl, 2023-04-08:
Use a modifier then and set different bottom layer values.

montagnon25, 2023-04-08:
Many thank's , it's resolve my bad missing habits.

Best regards

 

Philippe
 

Envoyé: samedi 8 avril 2023 à 13:55

closed 2023-04-25 by stepikovo