prusa3d/PrusaSlicer #10292: Make it possible to remember settings that I turned on

Lajbror, 2023-04-08:
**Is your feature request related to a problem? Please describe.**
There is no problem, it would just be nice if it was possible for the slicer to remember some settings that I turned on. For example if I turn on showing seems, it would be nice it the slicer remembers it until I tun it off.
So every time I start PrusaSlicer I would like to have showing seems enabled.

**Describe the solution you'd like**
Every time I start PrusaSlicer I would like to have showing seems enabled if I choose so.

**Describe how it would work**
If I change displaying any settings in the slicer, for example seems, retractions etc, I would like it to stay changed until I change it again even after I start the program again.

**Describe alternatives you've considered**

**Additional context**

neophyl, 2023-04-08:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/9468 or https://github.com/prusa3d/PrusaSlicer/issues/9706 etc

stepikovo, 2023-04-25:
I'm closing this as a mentioned duplicate. Thx @neophyl