prusa3d/PrusaSlicer #10303: Wait for to start - Bed probe being triggered before temp is set

DisasterofPuppets, 2023-04-09:
### Description of the bug

When I start a print with a Cr-10s machine I see this screen when I start a print. 

https://imgur.com/undefined

This starts probing before anything is heated, even though I have this set in the Gcode:
Issue does not occur with previous Versions of the software.

Start Gcode is set as below:

```
G90 ; use absolute coordinates
M83 ; extruder relative mode
G28 ; home all axis
M140 S60 ; set final bed temp
M104 S210 ; set final nozzle temp
M190 S60 ; wait for bed temp to stabilize
M109 S210 ; wait for nozzle temp to stabilize
G29; Level bed probe
G1 Z50 F240
G1 X2.0 Y10 F3000
G1 Z0.28 F240
G92 E0
G1 X2.0 Y140 E10 F1500 ; prime the nozzle
G1 X2.3 Y140 F5000
G92 E0
G1 X2.3 Y10 E10 F1200 ; prime the nozzle
G92 E0
G21 ; set units to millimeters
G90 ; use absolute coordinates
M83 ; use relative distances for extrusion
; Filament gcode
```

I have attached the full gcode, image and project file

### Project file & How to reproduce

[fanclip1.zip](https://github.com/prusa3d/PrusaSlicer/files/11184554/fanclip1.zip)
[image]
[Cliptest.zip](https://github.com/prusa3d/PrusaSlicer/files/11184555/Cliptest.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0-alpha6

### Operating system

Windows 11

### Printer model

Cr-10s

J3r0github, 2023-04-10:
So, the start gcode and the gcode file start seem to be identical, if we don't count comments and 
```
M201 X500 Y500 Z100 E5000 ; sets maximum accelerations, mm/sec^2
M203 X500 Y500 Z10 E60 ; sets maximum feedrates, mm / sec
M204 S500 T1000 ; sets acceleration (S) and retract acceleration (R), mm/sec^2
M205 X8.00 Y8.00 Z0.40 E5.00 ; sets the jerk limits, mm/sec
M205 S0 T0 ; sets the minimum extruding and travel feed rate, mm/sec
```
Which are not commands that could start probing?
No idea what "https://imgur.com/undefined" is but I know it sure isn't an image.
Now I might be stupid, but I am not able to reproduce this.

DisasterofPuppets, 2023-04-10:
Cheers, Will test and report back. (I updated the image link too)

closed 2023-04-12 by DisasterofPuppets