prusa3d/PrusaSlicer #10307: "Detect Bridging Perimeters" prints some perimeters at travel speed

kieraneglin, 2023-04-09:
### Description of the bug

## Context

I'm printing a model with unsupported overhangs. This isn't an issue, because I'm pausing the print before the overhangs and adding in other parts before I resume printing. These parts seem like they're completely unsupported, but in reality they will be printing on an inserted part rather than thin air.

## Expected Behaviour

When printing a model with unsupported overhangs and "Detect Bridging Perimeters", all perimeters should respect the speeds set in the "Speed for print moves" section.

## Actual Behaviour

When printing a model with unsupported overhangs and "Detect Bridging Perimeters", some perimeters are printed at your specified travel speed rather than any speed defined in the "Speed for print moves" section. This only occurs with "Detect Bridging Perimeters" enabled.

In the screenshots below, you'll see the gcode preview with my travels set to 150mm/s, 40mm/s, and 150mm/s but without "Detect Bridging Perimeters".

## Screenshots

### 150mm/s travel with "Detect Bridging Perimeters"
[image]

### 40mm/s travel with "Detect Bridging Perimeters"
[image]

### 150mm/s travel WITHOUT "Detect Bridging Perimeters"
[image]

### Project file & How to reproduce

[GoProMount_Overhang_Issue.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/11186123/GoProMount_Overhang_Issue.3mf.zip)

**This is the STL, just rename the `.txt` to `.stl`:** [Bell_Super_DH_-_GoPro_Mount v4.txt](https://github.com/prusa3d/PrusaSlicer/files/11186124/Bell_Super_DH_-_GoPro_Mount.v4.txt)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.6.0-alpha6+MacOS-arm64 | Also happened in Alpha 5 but I cannot confirm if it worked before then

### Operating system

MacOS Version 13.2.1 (22D68)

### Printer model

Heavily modified Ender 3 V2

rtyr, 2023-04-09:
Similar to https://github.com/prusa3d/PrusaSlicer/issues/10229

Jan-Soustruznik, 2023-04-12:
Confirmed, Issue closed as duplicate.

closed 2023-04-12 by Jan-Soustruznik