prusa3d/PrusaSlicer #10311: Oozing Of Foaming Filament / De-Retraction / Extra Length On Restart / Time Dependent

XCTracer, 2023-04-10:
When using foaming filament then the nozzle ooses filament when the extruder is not feeding any filament. The longer the extruder is idle the more foaming filament oozes out of the nozzle. 

The problem is now that when the extruder starts to extrude again that there is then not enough foaming filament inside the nozzle. So after a longer travel with an idle extruder the nozzle has lost a considerable amount of filament. Now when printing again then there is not enough filament coming out of the nozzle, and this leads to issues up to holes in the printed part. You can clearly see this in the provided pictures.

"extra lenght on restart" does not solve the issue. It may be too much or too little, depending on many milliseconds the extruder was not extruding.

Best would be to have an option for foaming filaments to define an "extra lenght on restart", depending on the time that the extruder has previously been on idle. A small look up table, user definable, would be good enough to solve the issue. The user can define a few lines, each line has an entry for how long the extruder was idle, and how much "extra lenght on restart" to add:

0ms / 0mm
200ms / 0.2mm
500ms / 0.4mm
1000ms / 0.7mm
2000ms / 1.2mm

It would be very easy to implement in the code: Start a timer when the extruder goes from extruding to idle, then calculate the dt when the extruder starts extruding again, and then use the small look up table to calculate the extra length on restart. With this simple function a lot of issues due to oozing (foaming) filament can get solved.

What would also be great is an option in PrusaSlicer to display how long the extruder is idle when the print head is moving without extruding. Like this the user can then check where the issue is, see how long the travel takes, and then modify the settings in the look up table accordingly. 

[image]
[image]
[image]
[image]
[image]

I have tried many different solutions, but nothing works. The longer the extruder is idle the more foaming filaments leaks out of the nozzle. Retraction does not help to fix the issue. Especially with an all metal hot end the retraction is limited. 

With "avoid crossing perimeters" the noozle ooses on the print itself, so that's not an issue. Just the missing filament when the extruder starts to extrude cannot get solved. All the tests have been made using PrusaSlicer 2.6.0-Alpha6.

DVSVIDEO, 2023-04-12:
Have you tried adding an extra length of retraction to this foaming material profile?
If you pull the material out of the hot zone no more gas develops to foam and push the material out...the usual retraction is not enough here.
I haven't tried it myself, it's just a logical question...

HB-Stratos, 2024-06-18:
Just chiming in to also say I would like to see this implemented. It would be super useful for LW-PLA as it is entirely impossible to stop it from oozing due to its foaming nature.
I believe the best implementation would be to measure the time between retraction and restart. I don't think going by travel distance has much merit as the oozing out of the nozzle is not (or only barely) affected by travel speed, but strongly affected by the time spent not extruding. 
I am also thinking of a Klipper Macro implementation of this feature, but I'm not sure if this is even possible yet, and if it was, it would be quite hacky. A slicer implementation would be best.

HB-Stratos, 2024-07-02:
Hey @XCTracer if you're still reading this. I have gone ahead and wrote a python gcode postprocessor that is capable of adding extra restart distance based on estimated travel time. It is very rough around the edges at the moment, but I could use help tuning it if you're interested. I've thrown the script on gist before I make it a proper repo, find it here:
https://gist.github.com/HB-Stratos/288777eaa019f8488e4804fa30b764e7

XCTracer, 2024-07-02:
I already did the same with a Python script. This works quite well, only little sand paper work is necessary afterwards. Send me an email at ***@***.*** for more information.

https://gist.github.com/HB-Stratos/288777eaa019f8488e4804fa30b764e7

HB-Stratos, 2024-07-02:
It looks like github automatically censored out your email

SabrCat, 2026-09-07:
I would still like to see this feature implemented.